MATPOWER/matpower #6: linprog and intlinprog parameters

clementwroyer, 2017-02-04:
Hi,

I just downloaded MATPOWER version 6.0 to use it together with MATLAB R2016b, on a MAC OS X Yosemite. When launching the testing script test_matpower, I experienced the following error:

***********************************************************

t_miqps_matpower........Error using intlinprog (line 132)
INTLINPROG has stopped because it encountered an internal error. We are sorry
for the inconvenience.

Please contact technical support for assistance with your problem, quoting the
code "-96@30".

Error in miqps_ot (line 281)
            intlinprog(c, intcon, Ai, bi, Ae, be, xmin, xmax, ot_opt);

********************************************************

After contacting MATLAB, I learnt that this issue is due to the parameter LPPreprocess in the options of initlinprog, which should be set to 'none' (instead of its default value 'basic') in order to avoid the error. I modified it before the call to intlinprog in the file miqps_ot.m, and it worked fine. I had the same issue (with the same error code) when linprog is called. In that case, the parameter to change is Preprocess (from 'basic' to 'none', as for LPPreprocess).

Having not seen previous messages regarding this error, I am wondering whether it is due to the latest  version of MATLAB (in the 2016b release, there was indeed a change on the default algorithm in linprog, see [https://www.mathworks.com/help/optim/release-notes.html](url)). I would also like to know whether you recommend using the version with the adjusted value of this Preprocess parameter or an older version with the default settings.

I hope this report will be helpful for the maintenance of this code, which I otherwise find very well organized and easy to use.

Thanks,

C. Royer

rdzman, 2017-02-06:
Back in March 2016, I encountered what appears to be a very similar bug (confirmed by MathWorks) in R2016a on Mac, and the workaround they gave me was the same. You'll notice that some of the tests for MOST (e.g. `t_most_suc.m`) include the line

      mpopt = mpoption(mpopt, 'intlinprog.LPPreprocess', 'none');

Unfortunately, adding that line to some other tests, actually triggered (rather than resolved) the INTLINPROG internal error, so I had to use it selectively in the tests.

However, on my machines (running macOS Sierra) I am not able to reproduce the problem you mention in `t_miqps_matpower`, so I'm not quite sure what to do.

```matlab
>> mpver                

MATPOWER               Version 6.0        16-Dec-2016
MATLAB                 Version 9.1        25-Aug-2016   Release: (R2016b)  
Optimization Toolbox   Version 7.5        25-Aug-2016   Release: (R2016b)  
MIPS                   Version 1.2.2      16-Dec-2016
MOST                   Version 1.0        16-Dec-2016
SDP_PF                 -- not installed --
YALMIP                 -- not installed --
BPMPD_MEX              -- not installed --
CLP                    -- not installed --
CPLEX                  -- not installed --
GLPK                   -- not installed --
Gurobi                 -- not installed --
IPOPT                  -- not installed --
KNITRO                 -- not installed --
MINOPF                 -- not installed --
MOSEK                  -- not installed --
PARDISO                -- not installed --
PDIPMOPF               -- not installed --
SCPDIPMOPF             -- not installed --
SDPT3                  -- not installed --
SeDuMi                 -- not installed --
TRALMOPF               -- not installed --
Architecture:          MACI64

  MATPOWER 6.0 is distributed under the 3-clause BSD License.
  Please see the LICENSE file for details.

>> t_miqps_matpower
1..240
skipped tests 1..48 : CPLEX not installed
skipped tests 49..96 : MOSEK not installed
skipped tests 97..144 : Gurobi not installed
skipped tests 145..192 : glpk not installed
ok 193 - intlin/lin/quadprog - 3-d LP : success
ok 194 - intlin/lin/quadprog - 3-d LP : x
ok 195 - intlin/lin/quadprog - 3-d LP : f
ok 196 - intlin/lin/quadprog - 3-d LP : lam.mu_l
ok 197 - intlin/lin/quadprog - 3-d LP : lam.mu_u
ok 198 - intlin/lin/quadprog - 3-d LP : lam.lower
ok 199 - intlin/lin/quadprog - 3-d LP : lam.upper
skipped tests 200..227 : intlin/lin/quadprog does not handle QP problems
ok 228 - intlin/lin/quadprog - infeasible LP : no success
ok 229 - intlin/lin/quadprog - 2-d MILP : success
ok 230 - intlin/lin/quadprog - 2-d MILP : x
ok 231 - intlin/lin/quadprog - 2-d MILP : lam.mu_l
ok 232 - intlin/lin/quadprog - 2-d MILP : lam.mu_u
ok 233 - intlin/lin/quadprog - 2-d MILP : lam.lower
ok 234 - intlin/lin/quadprog - 2-d MILP : lam.upper
skipped tests 235..240 : intlin/lin/quadprog does not handle MIQP problems
All tests successful (14 passed, 226 skipped of 240)
Elapsed time 0.05 seconds.
```

For a bug in 3rd party code, I hesitate to include a workaround for one setup that could potentially cause potentially cause problems for another when I can't test them all.

I guess my recommendation, until MathWorks fixes the bug, would be to use the latest version of Matlab and turn off the appropriate preprocess option only if and when you encounter the error.

And, of course, please do follow up with MathWorks so they eventually fix the bug.

clementwroyer, 2017-02-06:
Thanks for your answer, I will follow your recommendation. Regarding Mathworks, their only suggestion is to change the value of this parameter.

For your information, here is what I obtain without manually changing the value

```matlab
>> mpver           

MATPOWER               Version 6.0        16-Dec-2016
MATLAB                 Version 9.1        25-Aug-2016   Release: (R2016b)  
Optimization Toolbox   Version 7.5        25-Aug-2016   Release: (R2016b)  
MIPS                   Version 1.2.2      16-Dec-2016
MOST                   Version 1.0        16-Dec-2016
SDP_PF                 -- not installed --
YALMIP                 -- not installed --
BPMPD_MEX              -- not installed --
CLP                    -- not installed --
CPLEX                  -- not installed --
GLPK                   -- not installed --
Gurobi                 -- not installed --
IPOPT                  -- not installed --
KNITRO                 -- not installed --
MINOPF                 -- not installed --
MOSEK                  -- not installed --
PARDISO                -- not installed --
PDIPMOPF               -- not installed --
SCPDIPMOPF             -- not installed --
SDPT3                  -- not installed --
SeDuMi                 -- not installed --
TRALMOPF               -- not installed --
Architecture:          MACI64

  MATPOWER 6.0 is distributed under the 3-clause BSD License.
  Please see the LICENSE file for details.

>> t_miqps_matpower
1..240
skipped tests 1..48 : CPLEX not installed
skipped tests 49..96 : MOSEK not installed
skipped tests 97..144 : Gurobi not installed
skipped tests 145..192 : glpk not installed
ok 193 - intlin/lin/quadprog - 3-d LP : success
ok 194 - intlin/lin/quadprog - 3-d LP : x
ok 195 - intlin/lin/quadprog - 3-d LP : f
ok 196 - intlin/lin/quadprog - 3-d LP : lam.mu_l
ok 197 - intlin/lin/quadprog - 3-d LP : lam.mu_u
ok 198 - intlin/lin/quadprog - 3-d LP : lam.lower
ok 199 - intlin/lin/quadprog - 3-d LP : lam.upper
skipped tests 200..227 : intlin/lin/quadprog does not handle QP problems
ok 228 - intlin/lin/quadprog - infeasible LP : no success
Error using intlinprog (line 132)
INTLINPROG has stopped because it encountered an internal error. We are sorry
for the inconvenience.

Please contact technical support for assistance with your problem, quoting the
code "-96@30".

Error in miqps_ot (line 287)
            intlinprog(c, intcon, Ai, bi, Ae, be, xmin, xmax, ot_opt);

Error in miqps_matpower (line 220)
            miqps_ot(H, c, A, l, u, xmin, xmax, x0, vtype, opt);

Error in t_miqps_matpower (line 171)
        [x, f, s, out, lam] = `miqps_matpower(p); 
```
If I change this parameter's value, the output becomes identical to yours.

C. Royer

rdzman, 2017-02-06:
I hate non-deterministic computing 😜  ... but over the years I've learned not to expect absolutely identical behavior from Matlab OS versions, hardware platforms, etc, even with the same version of Matlab.

closed 2017-02-06 by rdzman