MATPOWER/matpower #9: some files in matpower6.0.zip can't be found in matpower-master

TheKiteRunning, 2017-04-29:
like `yalmip_options`

rdzman, 2017-05-01:
It is true. The `extras` directory (where `yalmip_options` is located) is not included in the core `matpower` repository. It has been moved to it's own repository at [`matpower-extras`](https://github.com/MATPOWER/matpower-extras). The plan is that these will still be included in future releases of a MATPOWER zip file, but the are going to be managed through a separate GitHub repository.

You simply need to place the contents of the `matpower-extras` repository in a directory named `extras` inside the MATPOWER install directory `<MATPOWER>` and re-run `install_matpower`.

TheKiteRunning, 2017-05-01:
Thank you for your kindly help.

By the way, a crash occurs when I run install_matpower. I have been racking my brain but still can't find a solution. 

Since the carsh_dump file is not supported to upload, the detail information is pasted below.

------------------------------------------------------------------------
          Access violation detected at Mon May 01 23:40:33 2017
------------------------------------------------------------------------

Configuration:
  Crash Decoding      : Disabled - No sandbox or build area path
  Crash Mode          : continue (default)
  Current Graphics Driver: Unknown hardware 
  Default Encoding    : GBK
  Deployed            : false
  Graphics card 1     : NVIDIA ( 0x10de ) NVIDIA GeForce 610M Version 9.18.13.2702
  Graphics card 2     : Intel Corporation ( 0x8086 ) Intel(R) HD Graphics 4000 Version 10.18.10.3308
  Host Name           : XX
  MATLAB Architecture : win64
  MATLAB Entitlement ID: Unknown
  MATLAB Root         : D:\Program Files\MATLAB\R2016b
  MATLAB Version      : 9.1.0.441655 (R2016b)
  OpenGL              : hardware
  Operating System    : Microsoft Windows 8.1 professional
  Processor ID        : x86 Family 6 Model 58 Stepping 9, GenuineIntel
  Virtual Machine     : Java 1.7.0_60-b19 with Oracle Corporation Java HotSpot(TM) 64-Bit Server VM mixed mode
  Window System       : Version 6.3 (Build 9600)

Fault Count: 1

Abnormal termination:
Access violation

Register State (from fault):
  RAX = 0000000000000001  RBX = 000000006a872860
  RCX = 000000018000d250  RDX = 000000009c2902c0
  RSP = 00000000040395a0  RBP = 00000000040396a0
  RSI = 0000000000000001  RDI = 0000000000000000
 
   R8 = 0000000004039688   R9 = 00007ffc49670000
  R10 = 0000000004039688  R11 = 0000000004039688
  R12 = 0000000000000000  R13 = 000000006a872860
  R14 = 0000000180000000  R15 = ffffffffffffffff
 
  RIP = 000000018000d360  EFL = 00010202
 
   CS = 0033   FS = 0053   GS = 002b

Stack Trace (from fault):
[  0] 0x000000018000d360 D:\Program Files\MATLAB\R2016b\bin\win64\libmx.dll+00054112 matrix::detail::noninlined::mx_array_api::mxIsContainer+00001136
[  1] 0x000000018000bcdf D:\Program Files\MATLAB\R2016b\bin\win64\libmx.dll+00048351 intrusive_ptr_release+00000303
[  2] 0x000000000af040e3 D:\Program Files\MATLAB\R2016b\bin\win64\libmwlxetypes.dll+00016611 MathWorks::lxe::ParseTypeScheme+00002011
[  3] 0x000000000af08944 D:\Program Files\MATLAB\R2016b\bin\win64\libmwlxetypes.dll+00035140 MathWorks::lxe::CreateXvalueOfMxArrayReference+00001444
[  4] 0x0000000008dace6f D:\Program Files\MATLAB\R2016b\bin\win64\m_lxe.dll+01560175 boost::serialization::singleton<boost::serialization::void_cast_detail::void_caster_primitive<ir::IrTree,MathWorks::utl::attachment> >::get_instance+00004915
[  5] 0x000000000af0b5c7 D:\Program Files\MATLAB\R2016b\bin\win64\libmwlxetypes.dll+00046535 MathWorks::lxe::LXETypeFactory::~LXETypeFactory+00000279
[  6] 0x0000000008d1555a D:\Program Files\MATLAB\R2016b\bin\win64\m_lxe.dll+00939354 boost::serialization::singleton<boost::archive::detail::iserializer<boost::archive::binaryTerm_iarchive,ir::IrTree> >::get_instance+00850278
[  7] 0x0000000008d16be3 D:\Program Files\MATLAB\R2016b\bin\win64\m_lxe.dll+00945123 boost::serialization::singleton<boost::archive::detail::iserializer<boost::archive::binaryTerm_iarchive,ir::IrTree> >::get_instance+00856047
[  8] 0x0000000008d19d39 D:\Program Files\MATLAB\R2016b\bin\win64\m_lxe.dll+00957753 boost::serialization::singleton<boost::archive::detail::iserializer<boost::archive::binaryTerm_iarchive,ir::IrTree> >::get_instance+00868677
[  9] 0x0000000008d1932f D:\Program Files\MATLAB\R2016b\bin\win64\m_lxe.dll+00955183 boost::serialization::singleton<boost::archive::detail::iserializer<boost::archive::binaryTerm_iarchive,ir::IrTree> >::get_instance+00866107
[ 10] 0x0000000008d1940d D:\Program Files\MATLAB\R2016b\bin\win64\m_lxe.dll+00955405 boost::serialization::singleton<boost::archive::detail::iserializer<boost::archive::binaryTerm_iarchive,ir::IrTree> >::get_instance+00866329
[ 11] 0x0000000008d08041 D:\Program Files\MATLAB\R2016b\bin\win64\m_lxe.dll+00884801 boost::serialization::singleton<boost::archive::detail::iserializer<boost::archive::binaryTerm_iarchive,ir::IrTree> >::get_instance+00795725
[ 12] 0x0000000008c89eab D:\Program Files\MATLAB\R2016b\bin\win64\m_lxe.dll+00368299 boost::serialization::singleton<boost::archive::detail::iserializer<boost::archive::binaryTerm_iarchive,ir::IrTree> >::get_instance+00279223
[ 13] 0x0000000008c89740 D:\Program Files\MATLAB\R2016b\bin\win64\m_lxe.dll+00366400 boost::serialization::singleton<boost::archive::detail::iserializer<boost::archive::binaryTerm_iarchive,ir::IrTree> >::get_instance+00277324
[ 14] 0x0000000008d11b9b D:\Program Files\MATLAB\R2016b\bin\win64\m_lxe.dll+00924571 boost::serialization::singleton<boost::archive::detail::iserializer<boost::archive::binaryTerm_iarchive,ir::IrTree> >::get_instance+00835495
[ 15] 0x0000000008d11953 D:\Program Files\MATLAB\R2016b\bin\win64\m_lxe.dll+00923987 boost::serialization::singleton<boost::archive::detail::iserializer<boost::archive::binaryTerm_iarchive,ir::IrTree> >::get_instance+00834911
[ 16] 0x0000000008d11979 D:\Program Files\MATLAB\R2016b\bin\win64\m_lxe.dll+00924025 boost::serialization::singleton<boost::archive::detail::iserializer<boost::archive::binaryTerm_iarchive,ir::IrTree> >::get_instance+00834949
[... cut ...]
[106] 0x00007ffc5b6313d2                   C:\Windows\system32\KERNEL32.DLL+00005074 BaseThreadInitThunk+00000034
[107] 0x00007ffc5d3d54e4                      C:\Windows\SYSTEM32\ntdll.dll+00087268 RtlUserThreadStart+00000052

If this problem is reproducible, please submit a Service Request via:
    http://www.mathworks.com/support/contact_us/

A technical support engineer might contact you with further information.

Thank you for your help.

选择文件 Try again with a PNG, GIF, JPG, DOCX, PPTX, XLSX, TXT, PDF, or ZIP.
 Styling with Markdown is supported
Comment Close and comment

rdzman, 2017-05-01:
The Matlab crash in `install_matpower` seems like a Matlab bug, but please open a separate issue for that.

closed 2017-05-01 by rdzman