MATPOWER/matpower #10: matlab crash when matpower toolbox is used

TheKiteRunning, 2017-05-02:
I really tried hard to solve this issue by my self, but fail again and again. 

1.  MATLAB crashes when the toolbox from github is used,  MATLAB and other toolboxes run normally.

When `mpver` runs, the result is:

MATPOWER               Version 6.1-dev    17-Mar-2017
MATLAB                 Version 9.1        25-Aug-2016   Release: (R2016b)  
Optimization Toolbox   Version 7.5        25-Aug-2016   Release: (R2016b)  
MIPS                   Version 1.3-dev    16-Mar-2017
MOST                   Version 1.0        16-Dec-2016
SDP_PF                 Version 1.0        17-Jan-2014
YALMIP                 Version                       
BPMPD_MEX              -- not installed --
CLP                    -- not installed --

then the crash occurs, the log is recorded in [the closed issue](https://github.com/MATPOWER/matpower/issues/9)

2. Even I delete the items in `set path` and reuse the zip files from [the website](http://www.pserc.cornell.edu/matpower/), the bug remains. 

Thanks a lot!

rdzman, 2017-05-02:
Based on where in the process the crash is occurring, I'm guessing that it is related to CPLEX. On my machine, CPLEX is not compatible with the latest versions of Matlab (since R2016a) and causes it to crash.

My suggestion is to either downgrade to Matlab R2016a or remove CPLEX from your Matlab path.

Of course, if that isn't the issue, please let me know. Maybe you can step through `mpver()` with the debugger to find the exact line at which the crash occurs.

TheKiteRunning, 2017-05-02:
2333, that's it! After the CPLEX folder is removed from the path, everything turns to normal.I should have checked it by myself. Thanks for your help!
-----
I find CPLEX works fine with R2016b, since I've run some scripts these days. What's more, after some tests, CPLEX and matpower 6.0 seem to coexist with each other peacefully. 

When the github version is used, if it doesn't notice the existence of CPLEX, then `mpver` reports no error, however `test_matpower` does:

>> test_matpower
t_test_fcns.................ok
t_nested_struct_copy........ok
t_feval_w_path..............ok
t_mpoption..................ok
t_loadcase..................ok
t_ext2int2ext...............ok
t_jacobian..................ok
t_hessian...................ok
t_margcost..................ok
t_totcost...................ok
t_modcost...................ok
t_hasPQcap..................ok
t_mplinsolve................ok (2 of 21 skipped)
t_mips......................ok
t_qps_matpower..............ok (216 of 360 skipped)
t_miqps_matpower............ok (178 of 240 skipped)
t_pf........................ok
t_pf_radial.................ok
t_cpf.......................ok
t_islands...................ok
t_opf_model.................ok
t_opf_fmincon...............ok
t_opf_mips..................ok (137 of 274 skipped)
t_opf_mips_sc...............ok (137 of 274 skipped)
t_opf_dc_gurobi.............ok
t_opf_dc_mips...............ok
t_opf_dc_mips_sc............ok
t_opf_dc_ot.................ok
t_opf_sdpopf................**Error using eig**
**Input to EIG must not contain NaN or Inf.**

Error in sdpopf_solver (line 1016)
        [evc, evl] = eig(double(A{i}));

Error in opf_execute (line 124)
      [results, success, raw] = sdpopf_solver(om, mpopt);

Error in opf (line 204)
[results, success, raw] = opf_execute(om, mpopt);

Error in runopf (line 75)
[r, success] = opf(casedata, mpopt);

Error in t_opf_sdpopf (line 61)
[baseMVA, bus, gen, gencost, branch, f, success, et] = runopf(casefile, mpopt);

Error in t_run_tests (line 61)
    feval( test_names{k}, ~verbose );

Error in test_matpower (line 140)
all_ok = t_run_tests( tests, verbose );

rdzman, 2017-05-02:
Ok, once again, this is a separate problem, so it deserves it's own entry in the issue tracker. I suspect you may have some but not all of the prerequisites for SDP_PF installed or something. If you want help tracking it down, create a new issue and please include the output of `mpver`.

closed 2017-05-03 by rdzman