MATPOWER/matpower #11: test_matpower error of MATPOWER 6.1-dev

TheKiteRunning, 2017-05-03:
Thanks for your patience and help all the time. The following is the previous procedure and error details:

- I've download a zip file of the MATPOWER repository from GitHub.
- I've changed the directory to <MATPOWER>.
- `install_matpower` has been executed
- `addpath` done
- `mpver`(CPLEX already removed from the MATLAB search path)

```
MATPOWER               Version 6.1-dev    17-Mar-2017
MATLAB                 Version 9.1        25-Aug-2016   Release: (R2016b)  
Optimization Toolbox   Version 7.5        25-Aug-2016   Release: (R2016b)  
MIPS                   Version 1.3-dev    16-Mar-2017
MOST                   Version 1.0        16-Dec-2016
SDP_PF                 Version 1.0        17-Jan-2014
YALMIP                 Version                       
BPMPD_MEX              -- not installed --
CLP                    -- not installed --
CPLEX                  -- not installed --
GLPK                   -- not installed --
Gurobi                 Version 7.0.2                 
IPOPT                  -- not installed --
KNITRO                 -- not installed --
MINOPF                 -- not installed --
MOSEK                  -- not installed --
PARDISO                -- not installed --
PDIPMOPF               -- not installed --
SCPDIPMOPF             -- not installed --
SDPT3                  -- not installed --
SeDuMi                 -- not installed --
TRALMOPF               -- not installed --
Architecture:          PCWIN64
```
- `test_matpower` reports error in `t_opf_sdpopf`

```
t_test_fcns.................ok
t_nested_struct_copy........ok
t_feval_w_path..............ok
t_mpoption..................ok
t_loadcase..................ok
t_ext2int2ext...............ok
t_jacobian..................ok
t_hessian...................ok
t_margcost..................ok
t_totcost...................ok
t_modcost...................ok
t_hasPQcap..................ok
t_mplinsolve................ok (2 of 21 skipped)
t_mips......................ok
t_qps_matpower..............ok (216 of 360 skipped)
t_miqps_matpower............ok (178 of 240 skipped)
t_pf........................ok
t_pf_radial.................ok
t_cpf.......................ok
t_islands...................ok
t_opf_model.................ok
t_opf_fmincon...............ok
t_opf_mips..................ok (137 of 274 skipped)
t_opf_mips_sc...............ok (137 of 274 skipped)
t_opf_dc_gurobi.............ok
t_opf_dc_mips...............ok
t_opf_dc_mips_sc............ok
t_opf_dc_ot.................ok
t_opf_sdpopf................Error using eig
Input to EIG must not contain NaN or Inf.

Error in sdpopf_solver (line 1016)
        [evc, evl] = eig(double(A{i}));

Error in opf_execute (line 124)
      [results, success, raw] = sdpopf_solver(om, mpopt);

Error in opf (line 204)
[results, success, raw] = opf_execute(om, mpopt);

Error in runopf (line 75)
[r, success] = opf(casedata, mpopt);

Error in t_opf_sdpopf (line 61)
[baseMVA, bus, gen, gencost, branch, f, success, et] = runopf(casefile, mpopt);

Error in t_run_tests (line 61)
    feval( test_names{k}, ~verbose );

Error in test_matpower (line 140)
all_ok = t_run_tests( tests, verbose );
```

rdzman, 2017-05-03:
It turns out that `test_matpower` only checks for the availability of SDP_PF and YALMIP before attempting to run `t_opf_sdpopf`, but doesn't check for the availability of an SDP solver.

I have updated `test_matpower` to also require that one of the following be available before attempting tests that involve calling YALMIP's `solvesdp`: MOSEK, SDP3, or SeDuMi.

Running `test_matpower` without an SDP solver should no longer produce a fatal error, but some of the SDP_PF tests may fail. I suggest you either install SeDuMi, if you think you might use SDP_PF functions, or otherwise remove SDP_PF from your Matlab path.

closed 2017-05-03 by rdzman

TheKiteRunning, 2017-05-04:
I get it, Thank you very much!

commit 0c2b503b (2017-05-03): Fix #11, fatal error from missing SDP solver.

- Fix fatal error encountered when running `test_matpower` with SDP_PF and YALMIP installed, but no SDP solver. Now checks for availability of SeDuMi, SDP3 or MOSEK before attempting to run SDP_PF tests that require solving an SDP.