MATPOWER/matpower #19: Dclines with negative losses

frederikgeth, 2017-07-27:
When defining a bidirectional dcline the line losses can be negative (e.g. for Loss0 = 0 and Loss1>0). It feels that a console warning / error / override is appropriate when defining such scenario.

rdzman, 2017-07-27:
What do you think of adding the following to `toggle_dcline`?

```Matlab
    l0 = mpc.dcline(:, c.LOSS0);
    l1 = mpc.dcline(:, c.LOSS1);
    k = find( (l0 + l1 .* mpc.dcline(:, c.PMIN) < 0) | ...
              (l0 + l1 .* mpc.dcline(:, c.PMAX) < 0) );
    if ~isempty(k)
        warning('toggle_dcline: loss can be negative for DC line from bus %d to %d\n', ...
            [mpc.dcline(k, c.F_BUS:c.T_BUS)]');
    end
```

frederikgeth, 2017-07-28:
Seems good, although I'm personally inclined to say`l1 .* mpc.dcline(:, c.PMIN) < 0)` is a sufficient  condition to give a warning. But then the warning should say that the line losses can be negative and/or lower than LOSS0. 

In the end it depends on how you look at the scope of the dcline model: 

- locally linearized in LOSS0 (losses lower than LOSS0 acceptable) 
- LOSS0 reflects a state-state consumption of the dcline's converter stations (losses lower than LOSS0 not acceptable)

rdzman, 2017-07-28:
Yeah, I was thinking of it as a linear approximation of the losses across the entire range from `PMIN` to `PMAX`, so losses less than `LOSS0` would be acceptable, but not negative losses.

To me checking that the total loss at each end of the range is positive, and therefore the loss is guaranteed to be positive for all feasible dispatch points, is the most straightforward and thorough, so I think that's what I'll include.

Thanks for the suggestion 👍

closed 2017-07-28 by rdzman

frederikgeth, 2017-07-28:
You're welcome.