MATPOWER/matpower #22: Issue regarding Installation

viswanath1494, 2017-09-04:
I tried installing MATPOWER on my computer. After adding it to the path in MATLAB, I typed in test_matpower and I got this error. How do I resolve this issue?
[image]

rdzman, 2017-09-05:
What version of MATPOWER is this? And did you use the installation instructions for your version? _(The GitHub version uses different paths and includes an installer. See [Installation](https://github.com/MATPOWER/matpower/#installation))_.

If that isn't the issue, please run the failing tests individually and provide the output here.
`t_feval_w_path`
`t_load_case`

rdzman, 2017-09-12:
Did you resolve this?

If we don't receive more info here in a few days, we will assume you were able to resolve the problem and will close this issue.

viswanath1494, 2017-09-23:
It got resolved. Thank you

closed 2017-09-23 by viswanath1494