MATPOWER/matpower #23: CPF termination problem

ElisNycander, 2017-09-06:
I have found what I believe to be a bug in the CPF function. For some systems, when close to the nose point, i.e. when the voltage starts falling steeply, instead of terminating the CPF will continue and voltages will instead start to increase rapidly. Although I have not investigated what causes this I believe it may be some numerical issue.

An example is provided here with the 4 bus system. The voltage at bus 3 falls to 0.55 and then starts increasing again. I have also observed the same behavior for more complicated systems. :

```
clear;
close all;

define_constants;

mpc = case4gs();

% change parameters
mpc.bus(4,[PD QD]) = [-1820 -1307.2];
mpc.gen(1,[PG QMAX QMIN PMAX]) = [300 200 -1e9 250];
mpc.gen(2,[QMAX PMAX]) = [1e9 500];

mpcb = mpc;
mpct = mpc;
% target case
mpct.bus(3,[PD QD]) = 2*mpcb.bus(3,[PD QD]);

% options
mpopt = mpoption();
mpopt.cpf.enforce_q_lims = 1;
mpopt.cpf.parameterization = 3;
mpopt.adapt_step = 0;
mpopt.step = 0.1;
mpopt.cpf.enforce_v_lims = 0;
mpopt.cpf.enforce_flow_lims = 0;
mpopt.cpf.plot.level = 2;

runcpf(mpcb,mpct,mpopt)
```

My matpower version:
```
MATPOWER               Version 6.1-dev    23-Aug-2017
MATLAB                 Version 8.6        13-Aug-2015   Release: (R2015b)  
Optimization Toolbox   Version 7.3        13-Aug-2015   Release: (R2015b)  
MIPS                   Version 1.3-dev    16-Mar-2017
MOST                   Version 1.0.1-dev  07-Aug-2017
SDP_PF                 -- not installed --
YALMIP                 -- not installed --
BPMPD_MEX              -- not installed --
CLP                    -- not installed --
CPLEX                  -- not installed --
GLPK                   -- not installed --
Gurobi                 -- not installed --
IPOPT                  -- not installed --
KNITRO                 -- not installed --
MINOPF                 -- not installed --
MOSEK                  -- not installed --
PARDISO                -- not installed --
PDIPMOPF               -- not installed --
SCPDIPMOPF             -- not installed --
SDPT3                  -- not installed --
SeDuMi                 -- not installed --
TRALMOPF               -- not installed --
Architecture:          PCWIN64
```

rdzman, 2017-09-06:
I haven't fully verified this, but if this is a bug, I'm pretty sure it falls in the category of unexpected behavior rather than incorrect behavior. You are enforcing Q limits, so when a Q limit is reached just before getting to the nose point, bus 4 is converted to a PQ bus. When that happens, the equations defining the nose curve (i.e. the nose curve itself) changes. If you set `mpopt.cpf.stop_at` to `'FULL'`, you can see that it traces out the top part of the new nose curve.

So the question here is, when making changes to the nose curve we are tracing (by changing bus types, etc.), how do we choose the direction to continue tracing on the new curve? I would need to look at the details of the code again, but I suspect that it is currently determined by the direction of change in lambda immediately preceding the change.

Is there a better way to select the direction?

@abhyshr, do you have any opinions here?

abhyshr, 2017-09-07:
@ElisNycander : Thank you for bringing up this interesting case and providing the test data! What you are observing here is the stable operating point transitioning to an unstable operating point when Q limit is encountered. At the base-case solution, the operating point is stable and as \lambda increases the point moves on the stable manifold (upper part of PV-curve with voltages decreasing) till it gen. 4 encounters the Qmax limit. At this point, when its bus-type get changed from PV to PQ, the operating point transitions to the unstable manifold (lower part of PV curve, but note that this is different PV curve). 

What you see after this, increase in the bus voltages, is a result of the tangent direction which is set to increase in the positive lambda direction. As lambda increases on the lower PV curve portion, the bus voltages increase, getting to the nose point and then turning back to further increase of the voltages. This also shows that a better voltage profile can be obtained if generator 4 is Q-limited. This is intuitively opposite but I think the reason for this is the large negative load (positive injection) at bus 4. 

So, now the question @rdzman is should we allow CPF to progress in the direction of lambda increase when a stable point transitions onto the unstable manifold or do we change the tangent direction and progress to lambda=0? Whether a point is on the stable or unstable manifold can be found out by checking the sign of the minimum eigen value of the Jacobian. For stable portion, all eigen values of Jacobian are positive and for unstable porition there is one negative eigen value. 

We could either choose one by default or have an option that lets the user choose how they want to proceed.

Sorry for the delay in replying. 

Thoughts?

rdzman, 2017-09-07:
Thanks @abhyshr. As a default, it would seem to make sense to me to always move along stable manifolds in the direction of increasing lambda and along unstable manifolds in the direction of decreasing lambda. I would guess this is the behavior most people would expect. I don't have a strong opinion on whether it's worth putting in a user option to control this. I think I'd leave it out unless someone thinks it's a good idea.

In any case, I suppose that in theory this switching from stable to unstable manifolds could also go the opposite direction, from unstable to stable. And it could be triggered by any event that causes a change in the system data (i.e. a change in the continuation curve being traced), not just by a Q limit. So we'll have to include that check for multiple cases.

Finally, what's the most efficient way to check the sign of the minimum eigen value of the Jacobian in Matlab?

abhyshr, 2017-09-07:
`Thanks @abhyshr. As a default, it would seem to make sense to me to always move along stable manifolds in the direction of increasing lambda and along unstable manifolds in the direction of decreasing lambda. I would guess this is the behavior most people would expect. I don't have a strong opinion on whether it's worth putting in a user option to control this. I think I'd leave it out unless someone thinks it's a good idea.`
I'm not sure CPF should be restricted to moving along the negative lambda direction for unstable manifolds. This would cause losing out on the information that switching actually is beneficial, and that the steady-state loading limit is improved by the reactive power limit. Perhaps, this information could be helpful for some planning folks?

Thinking a little more on this, I am wondering whether we should try to find the corresponding stable manifold solution after switching. Currently, on the unstable manifold, the solution after switching lies on the unstable manifold because, presumably, of the initial guess to Newton which is close to the unstable manifold. If we use a flat start initial guess then perhaps we can recover the stable manifold solution. 

`In any case, I suppose that in theory this switching from stable to unstable manifolds could also go the opposite direction, from unstable to stable. And it could be triggered by any event that causes a change in the system data (i.e. a change in the continuation curve being traced), not just by a Q limit. So we'll have to include that check for multiple cases.`
Yes, that's right. There can be a situation where the operating point transitions from unstable to stable manifold due to switching, for e.g. a cap bank being switched on due to low voltage.

`Finally, what's the most efficient way to check the sign of the minimum eigen value of the Jacobian in Matlab?`
`sign(min(real(eigs(J,1,'SR'))))` will give the sign of the eigen value of J with the smallest real part.

abhyshr, 2017-09-14:
Ray,
   Any thoughts on the comment I made on Github?

Shri

rdzman, 2017-09-15:
My first thought is that I like the continuity of the current approach, where the power flow solution is identical before and after the switch. That is, we are making a switch at the intersection of the two curves. I'm not sure what I think of the idea of using a flat start to "jump" to the stable manifold. Also, it seems to me that if the switch happens close to the nose point, it is conceivable that it wouldn't converge from a flat start. So I think my preference would be to simply select a direction to continue the trace from the switching point.

I guess my goal would be, first, make it behave in the most "expected" way by default and, second, as time permits include options for other types of behavior that might be useful.

What would you propose, @abhyshr?

jl-marin, 2017-09-21:
Sorry to arrive a bit late to the thread.  This is quite interesting; here's my take.  This issue is produced because there's a relatively unknown effect with PV-type buses: the fact that if the injection PG is too large (or, equivalently, if the setpoint VG is too low), the "high voltage" solution of the system when the bus is PV-type corresponds to a "low voltage" solution of the same system when the bus is type-switched to PQ.  This can be detected via the V-Q curve of the bus: these anomalous states are on the wrong side of this curve (i.e., negative sensitivity V vs. Q).

What to do with this in the context of a CPF curve?  I'm with Ray on employing the principle of "least surprise", and thus favor continuity of the the curve (anyway, the user will notice there's something wrong when the curve starts increasing instead of decreasing).  But Shri is right to point out that when this change is detected, maybe the user should be advised that the operating point could be corrected by re-adjusting the setpoint VG of said generator (one can calculate the new VG that would produce the same injection Q).  Maybe this could be calculated as an option.

Actually, if one wanted to be rigorous, the best thing would be to detect this problem **early**:  that is, detect the points lambda at which PV buses transition to the other side of their Q-V curve.  These transition points are actually nose points, but they are hidden because such tangency points are only bifurcation points in the PQ-type problem, not the PV-type problem (I mean, you'd only see them in the equations where the bus in question is treated as PQ).

JL

commit 3d1e9355 (2017-12-06): Final cleanup to fix #23 by adding logic to choose continuation direction after potential change from stable to unstable manifold.

(or vice versa)
closed 2017-12-06 by rdzman