MATPOWER/matpower #24: rate B and rate C not included in OPF

kknez, 2017-09-12:
Current MatPower version includes only branch rate A for OPF, and there is no automated way to include rate B or rate C even though they are defined in the MatPower case.

rdzman, 2017-09-12:
This should be quite simple and straightforward to implement. A good starting place for someone who would like to begin to contribute.

I propose adding an `opf.branch_rating` option with default value 'A', and alternate values 'B' and 'C'. If someone wants to tackle this, please indicate your intentions here to avoid duplication of effort.

mcsosa121, 2017-09-21:
If nobody has begun working on this, I would be interested in helping out.

rdzman, 2017-09-25:
👍  Go for it.

And may I suggest for the implementation that, rather than temporarily copying the values from the `RATE_B` or `RATE_C` columns into the `RATE_A` column of the `branch` matrix, you find all of the places in the code where `RATE_A` is used directly for the limit and select the appropriate column based on the `opf.branch_rating` option instead. Since this approach will not be possible for some of the OPF solvers (e.g. MINOPF, TSPOPF), for those cases you will have to go with the first approach.

Ideally, I'd also like to see new test cases added to the `t_opf_*` tests that verify the new option is working fine.

Feel free to ask here for input or feedback along the way. And thanks again for volunteering to help with this.

rdzman, 2018-01-26:
@mcsosa121 Are you working on this?