MATPOWER/matpower #26: CPF - issue with PQ/PV conversion with multiple generators at PV bus

ElisNycander, 2017-09-25:
I think there is a problem with converting PV buses with multiple generators to PQ buses in the CPF. What happens is that if one generator reaches its QMAX, the bus will be converted to a PQ bus even if there are other generators at the bus that still have more Q capacity. 

It happens in the following example using case4gs, where I have added an extra generator to bus 4. In the CPF, in the very first few steps, the QMAX = 167.2 of the new generator is reached, and bus 4 is converted to PQ. However, generator 1 is only at Q =-2.26 MVAr, i.e. not at its q limit. 

If instead an ordinary PF with the target case is run, the q limits are handled correctly, and both generators at bus 4 are at their maximum q limits, for a total of  367.8 MVAr. 

```` 
clear;

define_constants;

% options
mpopt = mpoption();

mpopt.out.all = 1;
mpopt.pf.enforce_q_lims = 1;

mpopt.cpf.enforce_q_lims = 1;
mpopt.cpf.parameterization = 3;
mpopt.adapt_step = 0;
mpopt.step = 0.1;
mpopt.cpf.enforce_v_lims = 0;
mpopt.cpf.enforce_flow_lims = 0;
mpopt.cpf.plot.level = 2;
mpopt.cpf.stop_at = 'FULL';

mpc = case4gs();

mpc.gen(1,[PG QMAX QMIN]) = [300 200 -1e9];
mpc.gen(2,QMAX) = 1e9;

% add extra generator
mpc.gen = [mpc.gen;
    4   200     0   167.8199 -167.8199 1.02 100 1 200 zeros(1,12)
];

mpcb = mpc;
mpct = mpc;
% increase load
mpct.bus(:,[PD QD]) = 2*mpct.bus(:,[PD QD]);

% run cpf
mpcc = runcpf(mpcb,mpct,mpopt);
% run pf
mpc2 = runpf(mpct,mpopt);
```` 
Matpower version:
````
MATPOWER               Version 6.1-dev    23-Aug-2017
MATLAB                 Version 8.6        13-Aug-2015   Release: (R2015b)  
Optimization Toolbox   Version 7.3        13-Aug-2015   Release: (R2015b)  
MIPS                   Version 1.3-dev    16-Mar-2017
MOST                   Version 1.0.1-dev  07-Aug-2017
SDP_PF                 -- not installed --
YALMIP                 -- not installed --
BPMPD_MEX              -- not installed --
CLP                    -- not installed --
CPLEX                  -- not installed --
GLPK                   -- not installed --
Gurobi                 -- not installed --
IPOPT                  -- not installed --
KNITRO                 -- not installed --
MINOPF                 -- not installed --
MOSEK                  -- not installed --
PARDISO                -- not installed --
PDIPMOPF               -- not installed --
SCPDIPMOPF             -- not installed --
SDPT3                  -- not installed --
SeDuMi                 -- not installed --
TRALMOPF               -- not installed --
Architecture:          PCWIN64
````

abhyshr, 2017-09-25:
Once again, thank you for raising the issue and providing a concrete example to test it.

I'll have to take a look at what's causing this issue.  I believe all the generators should hit their reactive power limits simultaneously. CPF uses the reactive power calculated by one of the core MATPOWER routines (I think it's pfsoln). MATPOWER uses a proportional sharing logic based on the reactive power range for dispatching generator reactive power. This results in all the generators incident at a bus hitting their Q limits simultaneously.

I will check and report back.

rdzman, 2017-12-05:
The issue here is related to generators at the same bus with wildly different Q ranges. The `pfsoln()` function distributes the total Q between generators at the same bus by choosing a factor _alpha_, between 0 and 1, such that the the sum of the _Qg_ across the generators at the bus is the total Q, and for each generator _Qg = Qmin + alpha * (Qmax - Qmin)_.

In the example provided, we have ...
Qmin1 = -1e-9, Qmax = 200
Qmin4 = -167.8199, Qmax4 = 167.8199

So, when the total Q is 185.32, _alpha_ is 0.9999998175, resulting in ...
Qg1 = 17.5025
Qg4 = 167.8199

So the upper bound on 4 appears to be binding, even though the upper bound on 1 is clearly not yet and won't be until _alpha_ gets extremely close to 1.

The solution (which I haven't yet implemented) is to check the `cpf.q_lims_tol` on the difference between the dispatch and the limit for the **totals** at the bus, not for the individual generators.

The temporary workaround is to avoid using generator Q limits with such wildly different ranges.

abhyshr, 2017-12-05:
Has the generator reactive distribution logic changed recently? I thought the generator reactive powers are distributed proportional to their reactive power range such that all the generators hit their reactive power limit simultaneously.

rdzman, 2017-12-05:
Well, it has changed recently, but only in the handling of gens with infinite ranges, which doesn't affect this case.

What I described above *is* proportional and, if you have infinite precision, they do reach their limits simultaneously. It's just that with wildly different ranges, numerically one falls within the `cpf.q_lims_tol` long before the other does.

For example, in this case, we have two generators that are 99.99998175% of the way to Qmax, for one (with a range of ~300MVAr) that's easily within `cpf.q_lims_tol`, while for the other one (with a range of 1e9) it's not even close to `cpf.q_lims_tol` yet.

commit ceae9a3c (2017-12-06): Fix #26 where CPF Q limit could be detected in error.

- Triggered by multiple generators at the bus with wildly different reactive ranges.
closed 2017-12-06 by rdzman