MATPOWER/matpower #29: Issue with adding nonlinear cost constraint

monisharaju, 2017-11-01:
I'm trying to add a non-linear cost constraint : cost = x' H x+e'x+M
I've used the direct method to create field which I've attached in the file below.

[Cost_constraint.docx](https://github.com/MATPOWER/matpower/files/1435696/Cost_constraint.docx)

One of the errors I get is the following :

> Subscripted assignment dimension mismatch.
> Error in opt_model/eval_nln_constraint (line 89)
>                 dg(i1:iN, 1:size(dgk, 2)) = dgk;

When I debugged I found, size(dgk, 2)=0 and the om.(" ") fields still have values from the branch flow non-linear constraint. I think my code is correct. It'd be helpful if you could point-out my mistake(s) or find out why I'm not able to add a new non-linear constraint.

Matpower version:

>>mpver

MATPOWER               Version 6.1-dev    27-Sep-2017
MATLAB                 Version 9.2        16-Feb-2017   Release: (R2017a)  
Optimization Toolbox   Version 7.6        16-Feb-2017   Release: (R2017a)  
MIPS                   Version 1.3-dev    16-Mar-2017
MOST                   Version 1.0.1-dev  27-Sep-2017
SDP_PF                 -- not installed --
YALMIP                 -- not installed --
BPMPD_MEX              -- not installed --
CLP                    -- not installed --
CPLEX                  -- not installed --
GLPK                   -- not installed --
Gurobi                 -- not installed --
IPOPT                  -- not installed --
KNITRO                 -- not installed --
MINOPF                 -- not installed --
MOSEK                  -- not installed --
PARDISO                -- not installed --
PDIPMOPF               -- not installed --
SCPDIPMOPF             -- not installed --
SDPT3                  -- not installed --
SeDuMi                 -- not installed --
TRALMOPF               -- not installed --
Architecture:          GLNXA64

MATPOWER 6.1-dev is distributed under the 3-clause BSD License.
Please see the LICENSE file for details.

rdzman, 2017-11-02:
_(Btw, it seems the Cost_constraint.docx file you attached is corrupted, but I eventually found a way to extract the contents.)_

There were several issues, mostly related to the calling syntax of the constraint and Hessian evaluation functions. In particular, they are similar, but not identical to the functions expected by MIPS. The details of the differences can be found in footnotes 31 and 32 on page 75 of [this version of the MATPOWER User's Manual](https://github.com/MATPOWER/matpower/files/1438519/MATPOWER-manual.pdf).

-  The constraint function `nli_constraint` returns only 2 values (`h` and `dh`), not 4.
- The value of `dh` returned by `nli_constraint` is the _m_ x _n_ Jacobian, whereas MIPS would expect the _n_ x _m_ transpose of that.
- The Hessian function does not include `cost_mult` (since, unlike with MIPS, this is only for a constraint term, no cost terms are included).
- The `lam` or `lambda` parameter to the Hessian evaluation function is simply a vector of lambdas corresponding to these constraints, not a struct containing multipliers for the full set of both equality and inequality constraints.

So, the following code runs for me, though it doesn't converge, since the added constraints make the problem infeasible.

```matlab
mpc= loadcase('case9');
[H, e, M] = cost_equation(mpc);
M2 = M-45000;
mpc.user_constraints.nli = {{'cost_fcn', 1, 'nli_constraint', 'nli_constraint_hess', {}, {H,e,M}}};
r = runopf(mpc);
```

```matlab
function [H, e, M] = cost_equation(mpc)
ng  = size(mpc.gen, 1);
nb  = size(mpc.bus, 1);
nx  = 2*nb + 2*ng;
a   = mpc.gencost(:,5);
b   = mpc.gencost(:,6);
PG_idx  = (2*nb +1):(2*nb +ng);
Acost   = a * mpc.baseMVA^2;
H       = sparse(PG_idx, PG_idx, Acost, nx, nx);
e       = zeros(nx, 1);
e(PG_idx,1) = b*mpc.baseMVA;
M = sum(mpc.gencost(:,7));
end
```

```matlab
function [h, dh] = nli_constraint(x, H, e, M)
h = x'*H*x +e' *x +M
dh = (2*H*x+e)';
end
```

```matlab
function d2H = nli_constraint_hess(x, lam, H, e, M)
d2H = 2 * lam * H;
end
```

monisharaju, 2017-11-02:
I tried the exact code you sent and it still doesn't work for me. I have the following errors.

Undefined function or variable 'nli_constraint'.

Error in opf_setup>@(x)nli_constraint(x,nlc{6}{:})

Error in opt_model/eval_nln_constraint (line 81)
        [gk, dgk] = fcn(xx);    %% evaluate kth constraint and gradient

Error in opf_consfcn (line 54)
    [h, dh] = om.eval_nln_constraint(x, 0); %% inequalities (branch flow limits)

Error in mipsopf_solver>@(x)opf_consfcn(x,om,Ybus,Yf(il,:),Yt(il,:),mpopt,il) (line 118)
gh_fcn = @(x)opf_consfcn(x, om, Ybus, Yf(il,:), Yt(il,:), mpopt, il);

Error in mips (line 353)
    [hn, gn, dhn, dgn] = gh_fcn(x); %% nonlinear constraints

Error in mipsopf_solver (line 121)
  mips(f_fcn, x0, A, l, u, xmin, xmax, gh_fcn, hess_fcn, opt);

Error in opf_execute (line 80)
      [results, success, raw] = mipsopf_solver(om, mpopt);

Error in opf (line 208)
    [results, success, raw] = opf_execute(om, mpopt);

Error in runopf (line 75)
[r, success] = opf(casedata, mpopt);

Error in cost_constraint (line 5)
r = runopf(mpc);

closed 2017-11-02 by monisharaju

rdzman, 2017-11-02:
The functions have to be in separate files for Matlab to be able to find them.

monisharaju, 2017-11-03:
Thank you that works.