MATPOWER/matpower #30: have_fcn('ipopt') with ipopt ver 3.12.7 gives an error because constraints are not arrays

fva11111, 2017-11-23:
I use OPTI Toolbox for Windows

It seems in have_fcn it's better to use
str = evalc('qps_ipopt([],[1; 1],[1 1],[2],[2],[1; 1],[1; 1],[1; 1],struct(''verbose'', 2))');
instead of
str = evalc('qps_ipopt([],1,1,1,1,1,1,1,struct(''verbose'', 2))');

**The first str is**
'This is Ipopt version 3.12.7, running with linear solver ma57.
     
     
     *** Error using Ipopt Matlab interface: ***
     The constraints must return a real, dense, double precision vector.
     
     Exception of type: IpoptException in file "Unknown File" at line -1:
      Exception message: Unknown Exception caught in Ipopt
     
     ans =
     
          1
     
     '
The second str is
    'This is Ipopt version 3.12.7, running with linear solver ma57.
     
     All variables are fixed and constraint violation 0.000000e+00
        is below tolerance 1.000000e-08. Declaring success.
     
     EXIT: Optimal Solution Found.
     
     ans =
     
          1
          1
     
     '

fva11111, 2017-11-23:
hmm.. this wasn't the problem

If I don't check the existence of ipopt_auxdata, I solve opf without any problems. If I do check, I get have_fcn = 0 and corresponding problems

rdzman, 2017-11-27:
I'm pretty sure this is due to a missing `ipopt_auxdata.m` file. I didn't realize (or else forgot) that OPTI Toolbox does not install this file, which is part of the Ipopt Matlab interface and is required by MATPOWER.

Just to confirm, with an unmodified MATPOWER, when you type `mpver` in a new Matlab session, does it give you the following warning?

`Warning: Improper installation of IPOPT. Version 3.12.7 detected, but
IPOPT_AUXDATA.M is missing.`

You can find `ipopt_auxdata.m` [in the Ipopt source distribution](https://projects.coin-or.org/Ipopt/browser/trunk/Ipopt/contrib/MatlabInterface/ipopt_auxdata.m?format=txt). I have also submitted an issue (jonathancurrie/OPTI#6) requesting that it be included in OPTI Toolbox.

In any case, **thanks** for bringing this issue to my attention, including the fact that my current test for a working Ipopt results in an `*** Error using Ipopt Matlab interface: ***`. While it doesn't prevent the check from working, it certainly is not the intended result.

fva11111, 2017-11-27:
Initially by typing mpver I get
IPOPT                  -- not installed --
But If I comment checking for ipopt_auxdata.m I get
IPOPT                  Version 3.12.7  

Thank you for ipopt_auxdata.m. It is the last piece of a ipopt puzzle that was missed in mailing lists.

rdzman, 2017-11-27:
So initially it said ...

`IPOPT                  -- not installed --`

... but did it also have the warning about the missing `ipopt_auxdata.m`? It should have. If it didn't, I'd like to figure out why, if possible.

fva11111, 2017-11-27:
Yes, I added it.

rdzman, 2017-11-27:
👍

closed 2017-11-27 by rdzman