MATPOWER/matpower #34: Newton's method power flow did not converge in simple example

bacalfa, 2018-02-28:
Am I incorrectly setting up the problem? I'm trying to solve [this](http://www.mty.itesm.mx/etie/deptos/ie/profesores/jabaez/clases/e00888/flujos_potencia/ejemplo_flujos_uwaterloo.pdf) example.

```
clear all
close all
clc

%% MATPOWER Case Format : Version 2
mpc.version = '2';

%%-----  Power Flow Data  -----%%
%% system MVA base
mpc.baseMVA = 100;

%% bus data
%	bus_i	type	Pd      Qd	Gs	Bs	area	Vm      Va	baseKV	zone	Vmax	Vmin
mpc.bus = [
    1       3       0       0   0   0   1       1.05    0   100     1       1.1     0.9;
    2       1       400     250 0   0   1       0       0   100     1       1.1     0.9;
    3       2       -200    0   0   0   1       1.04    0   100     1       1.1     0.9;
];

%% generator data
%	bus	Pg	Qg	Qmax	Qmin	Vg	mBase	status	Pmax	Pmin	Pc1	Pc2	Qc1min	Qc1max	Qc2min	Qc2max	ramp_agc	ramp_10	ramp_30	ramp_q	apf
mpc.gen = [
    1   0   0   0       0       0   100     1       0       0       0   0   0       0       0       0       0           0       0       0       0;
    3   0   0   0       0       0   100     1       0       0       0   0   0       0       0       0       0           0       0       0       0;
];

%% branch data
%	fbus	tbus	r       x       b	rateA	rateB	rateC	ratio	angle	status	angmin	angmax
mpc.branch = [
    1       2       0.02    0.04    0   0       0       0       0       0       1       -360    360;
    1       3       0.01    0.03    0   0       0       0       0       0       1       -360    360;
    2       3       0.0125  0.025   0   0       0       0       0       0       1       -360    360;
];

%% run pf
mpopt = mpoption('pf.nr.max_it', 100);
results = runpf(mpc, mpopt)
```

Output from PF analysis:

```
...
Warning: Matrix is close to singular or badly scaled. Results may be inaccurate. RCOND = -1.000000e+00. 

Newton's method power flow did not converge in 100 iterations.
```

In addition, is there an easy way to initialize a nonlinear PF with the solution of a linear PF?

abhyshr, 2018-03-01:
A few of the entries were incorrect. I’ve updated them below.

Thanks,
Shri
Ph: (630) 252 0219
www.mcs.anl.gov/~abhyshr<http://www.mcs.anl.gov/~abhyshr>

    3       2       0    0   0   0   1       1.04    0   100     1       1.1     0.9;

];

    1   0   0   0       0       1.05   100     1       0       0       0   0   0       0       0       0       0           0       0       0       0;

    3   200   0   0       0     1.04   100     1       0       0       0   0   0       0       0       0       0           0       0       0       0;

];

bacalfa, 2018-03-01:
Thanks for the quick reply. I'm still getting the same error.

```
clear all
close all
clc

%% MATPOWER Case Format : Version 2
mpc.version = '2';

%%-----  Power Flow Data  -----%%
%% system MVA base
mpc.baseMVA = 100;

%% bus data
%	bus_i	type	Pd  Qd	Gs	Bs	area	Vm      Va	baseKV	zone	Vmax	Vmin
mpc.bus = [
    1       3       0   0   0   0   1       1.05    0   100     1       1.1     0.9;
    2       1       400 250 0   0   1       0       0   100     1       1.1     0.9;
    3       2       0   0   0   0   1       1.04    0   100     1       1.1     0.9;
];

%% generator data
%	bus	Pg	Qg	Qmax	Qmin	Vg      mBase	status	Pmax	Pmin	Pc1	Pc2	Qc1min	Qc1max	Qc2min	Qc2max	ramp_agc	ramp_10	ramp_30	ramp_q	apf
mpc.gen = [
    1   0   0   0       0       1.05    100     1       0       0       0   0   0       0       0       0       0           0       0       0       0;
    3   200 0   0       0       1.04    100     1       0       0       0   0   0       0       0       0       0           0       0       0       0;
];

%% branch data
%	fbus	tbus	r       x       b	rateA	rateB	rateC	ratio	angle	status	angmin	angmax
mpc.branch = [
    1       2       0.02    0.04    0   0       0       0       0       0       1       -360    360;
    1       3       0.01    0.03    0   0       0       0       0       0       1       -360    360;
    2       3       0.0125  0.025   0   0       0       0       0       0       1       -360    360;
];

%% run pf
mpopt = mpoption('pf.nr.max_it', 100);
results = runpf(mpc, mpopt)
```

abhyshr, 2018-03-01:
Vm for bus 2 in mpc.bus struct should not be zero. Set it to 1.0.

abhyshr, 2018-03-01:
This is a simple usage question. I suggest you post all such questions on the MATPOWER forum instead.

closed 2018-03-01 by bacalfa