MATPOWER/matpower #37: Calculate PTDF for large cases

spmacdonald, 2018-05-23:
For large cases, `makePTDF` will fail because it tries to invert a very large matrix.  I found this discussion on the mailing list which gave some hints on how to proceed:

https://www.mail-archive.com/matpower-l@cornell.edu/msg05580.html

However, I am having trouble understanding what you mean by factoring `B_dc` to efficiently solve for dx.  I was hoping you might be able to provide some additional guidance on how to proceed.

Here is what I have done so far:

```
% https://www.mail-archive.com/matpower-l@cornell.edu/msg05580.html

% Calculate the shift factors without inverting Z
% Here I consider a 7 bus case that comes with PowerWorld and I try to compute the PTDFs with matpower

define_constants;

mpc = loadcase ('B7OPF.m');

[Bbus, Bf, Pbusinj, Pfinj] = makeBdc(mpc);

% For debugging
Bbus = full(Bbus)

% Bp is reduced Bbus obtained by removing ref bus:
Bp = full(Bbus);
Bp(:,7) = [];
Bp(7,:) = [];

% For debugging
Z = inv(Bp)

% Calculate H(1,1) element wise (equal to H(1,1))
x1 = mpc.branch(1, BR_X)
(Z(1,1) - Z(2,1)) / x1

% Shift factors
H = makePTDF(mpc)

% Calculate PTDF without inverting
% This matches Z(:,1)
a = Bp \ [1,0,0,0,0,0]';
b = Bp \ [0,1,0,0,0,0]';

Z
% This is H(1,:)
(a - b) / x1

% How can I calculate H(:,1) efficiently?

```
[B7OPF.zip](https://github.com/MATPOWER/matpower/files/2032070/B7OPF.zip)

abhyshr, 2018-05-23:
Scott,
  Using the same notation from the link, you’ll need to do a LU factorization (https://en.wikipedia.org/wiki/LU_decomposition
) of the B_dc matrix with MATLAB’s lu function first. This will give a lower triangular matrix L and an upper triangular matrix U that you can use to solve the linear system. If B_dc*H = delta{P_dc} is the linear system you want to solve then instead of

H = inv(B_dc}*delta{P_dc}
you would do
[L,U] = lu(B_dc};
H = U\(L\ delta{P_dc})

Note: Using MATLAB’s inv() function is terrible and expensive, instead MATLAB recommends to use the \ operator which uses Gaussian elimination.

Looking at the makePTDF() code, I see that the inv() function was removed and \ operator used at some point. I don’t know if this change was made in response to the below post. Regardless, the \ operator internally does a LU factorization and reuses the LU matrices when doing H = B_dc \ delta{P_dc}. So, makePTDF is efficient as it can be.

I would be surprised if makePTDF fails due to matrix inversion. My guess is for very large systems (my guess is in the order of ~100k buses), it fails due to the memory requirements needed to store dense H matrix (roughly 8 gigs).

Hence, the post suggests to calculate the sensitivities only for a subset of buses.

How big is your network, in terms of nbuses?

Thanks,
Shri
Ph: (630) 252 0219
www.mcs.anl.gov/~abhyshr<http://www.mcs.anl.gov/~abhyshr>

https://www.mail-archive.com/matpower-l@cornell.edu/msg05580.html

% https://www.mail-archive.com/matpower-l@cornell.edu/msg05580.html

jl-marin, 2018-05-23:
I think Ray was just referring to performing the LU decomposition (or LDLt if Bp is symmetric) explicitly, instead of relying on MATLAB's backlash operator.  That way you can **store the matrix factors**, which enables you yo solve linear systems **repeatedly** (changing only the r.h.s.).

See an example in this post:  https://www.mail-archive.com/matpower-l@cornell.edu/msg05499.html
In that case the goal was a bit different, namely having more control over which method MATLAB would use to factorize the matrix.  Nevertheless, it shows how to do it.

If you want to get down to the nitty-gritty, I strongly recomend this paper:

   _"Algorithm 930: FACTORIZE: an object-oriented linear system solver for MATLAB", T. A. Davis, ACM Transactions on Mathematical Software, Vol 39, Issue 4, pp. 28:1 - 28:18, 2013._

(you can get a copy of this article inside SuiteSparse's distribution).

I can give you a hand if you get stuck in this.

Jose Luis

spmacdonald, 2018-05-23:
Thanks for the fast responses guys, I will take some time to study them.  I should have mentioned that I am using Octave.  There is probably some internal differences between how MATLAB and Octave implement the \ operator?  My case is the PJM FTR network model, ~18,000 buses.  By fail I mean does not return in a reasonable amount of time, I left it running for about 24 hours.

jl-marin, 2018-05-23:
For dense matrices both MATLAB and Octave use optimize BLAS/LAPACK routines, but for _sparse_ ones the implementations are very different.  Tim Davis's paper explains the different sparse solvers inside MATLAB.  I'm not sure about Octave, I think it only uses UMFPACK, and you may have to check if lu() is overloaded to call it automatically; otherwise you may have to call a specific funcion in order to invoke it. 

Jose Luis

spmacdonald, 2018-05-23:
If I understand you correctly, you are implying that MATLAB's implementation would solve for *all* PTDFs in a reasonable amount of time?  I am really only interested in a small subset of lines.  In PowerWorld you specify an injection/extraction bus or group of buses and PTDFs are calculated for each branch given that MW transaction.  This is what I am trying to reproduce in Matpower.  See screenshot.
[image]

rdzman, 2018-05-23:
Thanks Shri and Jose Luis for the good responses here. Your help answering questions is always appreciated.

Scott, I don't think Matlab and Octave should be fundamentally different. Both will take forever to compute the full PTDF (which in your case is a dense matrix of probably 3 GB or so).

If you want to compute the PTDF corresponding to a particular transfer (injection at bus _i_, extraction at bus _j_), you should be able to do it efficiently with something like:

```matlab
slack = find(mpc.bus(:, BUS_TYPE) == REF);
noslack = find(mpc.bus(:, BUS_TYPE) ~= REF);
[Bbus, Bf, Pbusinj, Pfinj] = makeBdc(mpc);
Bbus(slack, :) = [];   % remove slack row
Bbus(:, slack) = [];   % remove slack col
dP = zeros(nb, 1);
dP(i) = 1;
dP(j) = -1;
dP(slack) = [];   % remove slack
dx = Bbus \ dP;
dTheta = zeros(nb, 1);
dTheta(noslack) = dx;
H = Bf * dTheta;
```

If you want to compute for multiple transfers, as was said above, you can save time by using `lu()` and storing the factors, rather than using `\` every time. So you would do something like this once ...

```matlab
[L,U] = lu(Bbus);
```

Then for each `dP`, you'd do ...

```matlab
dx = U \ L \ dP;
```

spmacdonald, 2018-05-23:
Thank you!  That is exactly what I wanted.

jl-marin, 2018-05-23:
Let me add to Ray's answer that, if you're using Octave, you may want to call LU with at least **four outputs**, in order to have UMFPACK use a fill-reducing ordering.  From https://octave.sourceforge.io/octave/function/lu.html:

>  When called with two or three output arguments and a sparse input matrix, lu does not attempt to perform sparsity preserving column permutations. Called with a fourth output argument, the sparsity preserving column transformation Q is returned, such that P * A * Q = L * U.

If the matrix is very large, such reordering can make a huge performance difference.

Jose Luis

rdzman, 2018-05-24:
I almost mentioned that. I actually like to use the vector form, like this (from [here](https://github.com/MATPOWER/mips/blob/aebea8dde6aeec51496d69b5f0fce273ab85f748/lib/mplinsolve.m#L144)) for solving `A * x = b` ...
 ```matlab
[L, U, p, q] = lu(A, 'vector');
x = zeros(size(A, 1), 1);
x(q) = U \ ( L \ b(p));
```

closed 2018-06-04 by rdzman