MATPOWER/matpower #39: Problem with runopf

mariodra, 2018-06-02:
An error occurred while trying to determine whether "opf_model" is a function name.

This is the error message I get while either testing Matpower or running the opf. Have you got any clue what the cause might be?

mariodra, 2018-06-02:
Okay, apparently the problem occurs when 2 versions of Matpower are placed on the hard drive. It doesn't  matter if only one is added to Matlab's path

rdzman, 2018-06-04:
Glad you seem to have found a way to resolve the issue, though I don't think your diagnosis is quite accurate.

I have many versions of MATPOWER on my machine. The important thing really is to make sure that you don't have functions in your Matlab path that mask any of the MATPOWER functions (e.g. from other MATPOWER versions).

I do this by never adding MATPOWER permanently to my path, but having different scripts with the appropriate `addpath()` calls for the different MATPOWER installations. This has worked well for me for many years.

closed 2018-06-04 by rdzman