MATPOWER/matpower #44: matpower invalid call to length (psse2mpc)

cemcclelland, 2018-08-31:
Can't figure out why the following error is thrown when calling psse2mpc in command line and on test_matpower. 

t_psse..................error: Invalid call to length.  Correct usage is:

 -- length (A)
error: called from
    print_usage at line 91 column 5
    psse_parse_line at line 65 column 12
    t_psse at line 61 column 14
    t_run_tests at line 58 column 5
    test_matpower at line 128 column 1

rdzman, 2018-08-31:
Are you using Matlab or Octave?
What version?
What version of MATPOWER? 6.0? or the latest GitHub version?

cemcclelland, 2018-08-31:
Hi Ray- I'm running Octave on Windows 64, just downloaded / reinstalled
today. Also downloaded the latest MATPOWER today. Reinstalled Java too.

rdzman, 2018-08-31:
Latest MATPOWER from GitHub or from the MATPOWER home page?

rdzman, 2018-08-31:
I assume you're using MATPOWER 6.0 from the MATPOWER home page. Try getting the latest development version from GitHub. Be sure to follow the installation instructions from the GitHub page, as they have changed.

I think that should fix the problem for you. Let me know whether it does or not.

cemcclelland, 2018-09-01:
Ray, I downloaded the MATPOWER zip from Github and followed the
instructions.
Still, I'm getting:

Would you like to run the MATPOWER tests now? [Y or N] (default = N) : y
error: 't_run_tests' undefined near line 143 column 10
error: called from
    test_matpower at line 143 column 8
    install_matpower at line 376 column 13

cemcclelland, 2018-09-01:
Ray, disregard my last email. This time I got the directory right and the
test ran properly, seems i'm back in business.

Thank you!

closed 2018-09-01 by rdzman