MATPOWER/matpower #45: Problem in newtonpf.m

ywlgy, 2018-10-05:
Hi,

I am having a problem when using newtonpf.m alone. This problem exists in github version as well as in 6.0. Version 5.1 and before are fine.

Line 53: mis = V .* conj(Ybus * V) - Sbus(Vm);

Since Vm is non-integer, MATLAB will give an error message saying subscript cannot be a non-integer.

In the previous version, only Sbus is used.

Similar usage can be found in Line 88 and 115.

Can anyone explain this?

rdzman, 2018-10-05:
There was a change made between v5.1 and v6.0 that should have been flagged as an INCOMPATIBLE CHANGE in the release notes, but apparently got missed. If you are calling `newtonpf()` directly, rather than via `runpf()`, it is important to note that the `Sbus` input argument is no longer a vector, but rather it is a handle to a function that returns a vector of bus power injections given a vector of voltage magnitudes as an input. Unfortunately, it seems this is not documented clearly in the help text for the updated `newtonpf` either, so I'm going to leave this issue open until I get a chance to fix that.

See line 199 in `runpf.m` for an example of how the `Sbus` argument is defined.

rdzman, 2018-11-12:
Addressed in a9b8bf3 and 021d877.

commit 021d8774 (2018-11-12): Add missing notice of INCOMPATIBLE CHANGE in v6.0.

- Fix #45.
- Change Sbus arg in newtonpf() to function handle.

Refs: https://github.com/MATPOWER/matpower/commit/a9b8bf3a8b26f995e0f35011bf42eb5deb7b8211
closed 2018-11-12 by rdzman