MATPOWER/matpower #47: Nodal LMPs

spmacdonald, 2018-11-02:
Hi,

I am interested in calculating the components of the nodal LMPs (energy, loss, congestion).  After reading the documentation I understand that the `bus(i, LAM_P)` value represents the total LMP at bus i.  

I searched through the code for where this is calculated and I think the file `lib/fmincopf_solver.m` is where this value is set, is that correct?  It is not clear to me how I can calculate the individual components of the LMP, is this done anywhere in the code?

Thank you for your time and efforts.

rdzman, 2018-11-02:
The nodal prices in `bus(i, LAM_P)` are computed directly by the selected OPF solver as the shadow prices on the power balance constraints, so where/how constraint shadow prices are computed is specific to the individual solver.

MATPOWER does not currently include any code to break down these prices into individual components. It seems there are numerous ways to define the various components.

I would be very open to including contributed code to compute these components if you happen to have the inclination to implement that feature.

Here are a some relevant papers to consider (listed partly for my own reference), and I'm sure there are many others:

1. Luonan Chen, H. Suzuki, T. Wachi and Y. Shimura, ["Components of nodal prices for electric power systems,"][1] in IEEE Transactions on Power Systems, vol. 17, no. 1, pp. 41-49, Feb. 2002.
doi: 10.1109/59.982191

2. Xu Cheng and T. J. Overbye, ["An energy reference bus independent LMP decomposition algorithm,"][2] in IEEE Transactions on Power Systems, vol. 21, no. 3, pp. 1041-1049, Aug. 2006.
doi: 10.1109/TPWRS.2006.876703

3. T. Orfanogianni and G. Gross, ["A General Formulation for LMP Evaluation,"][3] in IEEE Transactions on Power Systems, vol. 22, no. 3, pp. 1163-1173, Aug. 2007.
doi: 10.1109/TPWRS.2007.901297

4. D. Cheverez-Gonzalez and C. L. DeMarco, ["Mutually orthogonal LMP decompositions: Congestion decomposes, losses do not,"][4] 2009 6th International Conference on the European Energy Market, Leuven, 2009, pp. 1-6.
doi: 10.1109/EEM.2009.5207186

5. F. Li, E. Litvinov and T. Zheng, ["A combined LMP model considering reference independent decomposition and fictitious nodal demand,"][5] IEEE PES General Meeting, Providence, RI, 2010, pp. 1-5.
doi: 10.1109/PES.2010.5589739

6. D. Chéverez-González and J. Lin, ["Mutually orthogonal LMP decomposition: Analysis of PJM network by null space approach,"][6] IEEE PES General Meeting, Providence, RI, 2010, pp. 1-8.
doi: 10.1109/PES.2010.5588186

7. V. Sarkar and S. A. Khaparde, ["Optimal LMP Decomposition for the ACOPF Calculation,"][7] in IEEE Transactions on Power Systems, vol. 26, no. 3, pp. 1714-1723, Aug. 2011.
doi: 10.1109/TPWRS.2011.2104371

8. P. Lipka, S. S. Oren, R. P. O’Neill and A. Castillo, ["Running a More Complete Market With the SLP-IV-ACOPF,"][8] in IEEE Transactions on Power Systems, vol. 32, no. 2, pp. 1139-1148, March 2017.
doi: 10.1109/TPWRS.2016.2569596

[1]: https://doi.org/10.1109/59.982191
[2]: https://doi.org/10.1109/TPWRS.2006.876703
[3]: https://doi.org/10.1109/TPWRS.2007.901297
[4]: https://doi.org/10.1109/EEM.2009.5207186
[5]: https://doi.org/10.1109/PES.2010.5589739
[6]: https://doi.org/10.1109/PES.2010.5588186
[7]: https://doi.org/10.1109/TPWRS.2011.2104371
[8]: https://doi.org/10.1109/TPWRS.2016.2569596

closed 2018-11-02 by rdzman

spmacdonald, 2018-11-02:
My understanding from reading the PJM and NYISO market documentation is that LMP is calculated as:

`Lambda_i = Lambda - LF_i * Lambda + sum_{k=1 to K} S_{i,k} * mu_k`

Where Lambda is the system energy price, LF_i is the loss factor for bus i, S_{i,k} are the shift factors for bus i and branch k and mu_k is the shadow price for branch k.

They define the Loss Factor as the sensitivity of system losses to a change in injection at the bus i.

I was trying to calculate this LF_i vector in a straight forward way using matpower by looping over the buses, adding 1 MW of injection and re-running the OPF and computing the difference in system losses.  Is this a reasonable approach?

In general what would you suggest as an approach to decomposing the LMP into components?  I think it makes sense to try and approximate what the ISOs are doing, do you agree?

I have much to learn still so apologies for inaccurate statements.

rdzman, 2018-11-05:
The point is that none of the values needed for the decomposition are direct results of the OPF optimization problem itself. And there are multiple ways to compute loss factors, shift factors, etc. many of which depend on the (somewhat arbitrary) choice of reference bus, etc. The direct results of the optimization are the nodal prices themselves and the branch shadow prices.

I do think it makes sense to try to approximate what the ISO's are doing if you need to decompose the nodal prices, but for that you need the details of how they compute the loss factors, shift factors, etc. I just haven't taken the time to look up all of those details and I wouldn't be surprised it varies by ISO. It's possible that references 5 and 6 in my previous post would describe the details of the methods used by ISONE and PJM, but I'm not sure.