MATPOWER/matpower #60: slack bus has no generator in RTE cases

rdzman, 2019-03-06:
It appears that the RTE cases have no generator at the slack/reference bus.

```
case1888rte
case1951rte
case2848rte
case2868rte
case6468rte
case6470rte
case6495rte
case6515rte
```

Currently, MATPOWER's `runpf` handles this by (silently) converting the specified slack/reference bus to a PQ bus, and selecting the first PV bus as a new slack/reference bus.

However, this should be corrected in the cases by specifying a slack/reference bus that has a generator.

_Thanks to Drosos Kourounis for pointing this out._

ccoffrin, 2019-03-07:
To the best of my knowledge, these cases originally appeared in this [technical report](https://arxiv.org/abs/1603.01533).  If the version distributed with Matpower changes from the original report, I worry that many people will publish papers without indicating which version they used, which will make it difficult to reproduce results down the road.

Thoughts?

rdzman, 2019-03-07:
Thanks. I share your hesitation about creating multiple versions of a case. On the other hand, the motivation to "fix" this is precisely for reproducibility of results.

At the moment, the power flow problem for each of these cases is not clearly defined, so any power flow solution is necessarily dependent on whatever convention was used by the particular power flow software internally to "fix" such the case before running it. MATPOWER's convention is undocumented and likely differs from that of other software. In fact, some software may simply treat this as an error and require that the case be fixed manually.

So, in my opinion, modifying these cases to use a valid slack bus is a minor change and the problem is serious enough to warrant a new "fixed" version. What do you think?

I'm in contact with the Cédric and his RTE colleagues about the issue.

ccoffrin, 2019-03-07:
I think the best solution would be to fix both versions.  So if the RTE team agrees to update the technical report along with the updates you make here, problem solved!

rdzman, 2019-03-07:
Great idea. I'll mention that to them. Thanks.

rdzman, 2019-03-25:
According to Jean Maeght of RTE-France, if I understand correctly, their operational power flow software uses the most meshed 400kV bus (which does not have a generator) as a temporary slack bus while iteratively solving the problem and distributing whatever slack arises among all generators.

My suggestions are:
1. Simply select a 400kV bus that already has a generator (maybe the largest one?) as the slack bus. If the active power output of the other generators is a power flow solution (i.e. the current slack is zero), changing the slack bus for MATPOWER as suggested should not change the power flow solution.
2. Give each of the updated RTE cases an explicit version number that appears in the comments at the top of the file and in an updated to the [original arXiv paper](
https://arxiv.org/abs/1603.01533) in which the data was first published. This version number could either be numerical (e.g. v2) or a date (e.g. v2019-03-25), but it would give researchers a way to specify explicitly which version they are using.

What does everyone think?

jpjmaeght, 2019-06-05:
Hello,
In our industrial power flow software (named Hades, available here http://www.itesla-pst.org/ and http://www.rte.itesla-pst.org/ ), slack bus is in fact distributed.
We first look for the most meshed 400kV bus, and we use it as slack bus for the first Gauss-Newton run.
After Gauss-Newton convergence, the slack value at slack bus is dispatched over all generating units, proportionally to their participation to primary frequency reserve (or proportionally to their distance to Pmax in some cases).
Then a Gauss-Newton method is run again.
Then slack value is dispatched again.
And so on. A few iterations only are necessary until the slack value is zero.
So in that sense our slack bus is distributed.

In case of OPF computation, slack bus has no sense. In case of OPF, one might want to fix phase=0 for one bus (although this is not strictly necessary). Often people take slack bus for phase=0, but there is no real reason for. Smart choice of "phase=0" bus should be a research topic. Relying on "phase=0" bus in data sets is not a good choice.

I am now modifying our cases to add a generator on slack bus.
kind regards
Jean

jpjmaeght, 2019-06-05:
After adding a generator at slack bus, I will modify the comments in case files, in order to mention that this is a new version.
My proposal would be to write "version 2019".
After we agree on this, I will also change the files in our arxiv paper.
Jean

rdzman, 2019-06-05:
Thanks. I think "version 2019" is ok, but it arbitrarily eliminates the possibility of more than 1 update in a given year. Hopefully, that will never happen, but I'd prefer either something like v2 or v2019-06-05. I'll leave the final call up to you.

commit ad48e305 (2019-06-05): Fix #60 by adding missing generator at slack bus in RTE cases

Affects the following cases:
    - `case1888rte`
    - `case1951rte`
    - `case2848rte`
    - `case2868rte`
    - `case6468rte`
    - `case6470rte`
    - `case6495rte`
    - `case6515rte`
- Thanks to Jean Maeght.

Refs: https://github.com/MATPOWER/matpower/issues/60
closed 2019-06-05 by rdzman