MATPOWER/matpower #63: clarify slack distribution among multiple gens at slack bus

rdzman, 2019-03-11:
Currently, when there are multiple generators at the slack bus, MATPOWER's power flow considers the first generator at the bus to be the slack generator, taking up all of the slack. However, I'm not sure if this convention is documented anywhere and it is probably not the behavior intuitively expected by most users.

I propose to update MATPOWER's handling of slack allocation among multiple generators at a slack bus in the following way:
1. If area participation factors (`APF` column of `gen` matrix) are provided, split the slack according to the proportions indicated by the relative participation factors.
2. If area participation factors are not provided (i.e. set to 0), the slack will be distributed according to the size of the generator. I.e. for identical generators it will be distributed evenly.

This also needs to be mentioned explicitly in the documentation.

rwl, 2019-03-13:
The output of the slack generator often serves as a convenient way to quickly compare power flow results. For example, in [Andes](https://github.com/cuihantao/Andes) there is an effort to use the output from MATPOWER, via [Oct2PYPOWER](https://github.com/rwl/oct2pypower), to verify power flow results. Both of the suggestions made seem to be worthwhile improvements over the current approach. However, given that this would represent a breaking change for MATPOWER v7 it might be worth starting a thread on the mailing list so that this could be discussed further with the user community.

rdzman, 2019-03-13:
Good point. I'll just mention that the motivation for this change is to remove ambiguity in the handling of slack for such cases, in order to enhance the reproducibility of results.

Feel free to discuss on the mailing list (see [this thread](https://www.mail-archive.com/matpower-l@cornell.edu/msg06780.html)), or here.