MATPOWER/matpower #64: distribution power flow

Jerrywlj, 2019-04-22:
The suggestion I would like to make is about the power flow calculation module of distribution network. Although the module has only one requirement for node numbers, many cases cannot be met. I am doing a research on distribution network reconfiguration, in which the power flow calculation should not have any other node requirements except the power supply node.

todorovski-m, 2019-04-22:
If I have understood you correctly you have a concern about bus ordering in distribution power flow methods. All three of them calc_v_pq_sum, calc_v_i_sum and calc_v_y_sum require specific ordering in order to apply backward-forward summation and voltage calculation. Such bus ordering is obtained with order_radial which is called at the very beginning of radial_pf. Function radial_pf requires that MATPOWER case struct is with internal bus numbering, which is also the case since at the beginning of run_pf there is a call to ext2int that performs external to internal indexing.
In such a way you may have any bus numbering scheme in your case file and MATPOWER will take care for all bus renumbering and at the end will give the results for the original bus numbering.

Jerrywlj, 2019-04-23:
> If I have understood you correctly you have a concern about bus ordering in distribution power flow methods. All three of them calc_v_pq_sum, calc_v_i_sum and calc_v_y_sum require specific ordering [...]

Thank you very much for your reply. What I need is just the power flow calculation of the distribution network that meets the method of forward extrapolation with any number.Before that, I did not read the code carefully and applied this module, but just read the user manual, which described that the sending bus number should be smaller than the receiving bus number. After reading the code, I found more information about the node number.

akhilazm, 2019-05-27:
in "case33bw" file the tieswitches are provided with BR_STATUS as 0 as they are initial out of service. but I am not getting the other small distribution system case file with the tieswiitches. are these network compatible with network reduction?

Jerrywlj, 2019-05-28:
There is only one' case33bw' in matpower that can be used for network reduction. If you plan to use other data for network reduction, you can look for PG&E 69 node distribution network data.

rdzman, 2019-05-29:
I believe the original poster's question was addressed, so I'm closing this. If you have other questions, please open another issue or ask on the [MATPOWER-L mailing list](https://www.pserc.cornell.edu/matpower/mailinglists.html#discusslist).

closed 2019-05-29 by rdzman