MATPOWER/matpower #67: After compile the matpower, the IPOPT can not be found

shiftlin89, 2019-06-19:
Usually, I use Matlab GUI to run matpower code.  Call the IPOPT solver(ipopt.mexa64) to solve the problems, it works perfectly.
However, Recently I need to compile it to be an executable program. I use the Matlab compile toolbox to generate the executable file. When I try to run the program in the cmd window,  It comes the error: "Error using opf_execute (line 86) opf_execute: MPOPT.opf.ac.solver = 'IPOPT' requires IPOPT (see http://www.coin-or.org/projects/Ipopt.xml)" i.e. It can not find the IPOPT solver, I did include the IPOPT files in the setpath interface.

I tried a lot about this problem but still can not solve it, does anyone can help me?

rdzman, 2019-06-20:
I do not have experience with compiling MATPOWER, or using the MATLAB compiler at all, for that matter. I'm not sure if it's a problem with linking in the IPOPT MEX file or something related to how the compiler handles calls to `evalc()` which is used in `have_fcn()` to detect IPOPT.

In order to track down the problem, one suggestion would be to try compiling a simple test script that includes only a few of the commands that are used inside `have_fcn()` to detect the presence of IPOPT.

For example, try compiling a simple script that contains ...

```
fprintf('exist returns: %d\n', exist('ipopt', 'file'));
str = evalc('qps_ipopt([],[1; 1],[1 1],[2],[2],[1; 1],[1; 1],[1; 1],struct(''verbose'', 2))');
fprintf('running qps_ipopt outputs: %s\n', str);
```

Run it directly in MATLAB first, then in the compiled script to see the difference.

shiftlin89, 2019-06-21:
Dr. Zimmerm, appreciate for your advice.  I have solved this problem now. Here is my experience.

I did some test,  for the example you given and Matpower version 6.0, after compiled, the executable program can find IPOPT and works well.  However, for the Matpower version 7.0 , the compiled program failed.

Then. my tricky method is:  I integrate your example into the Matpower 7.0 , using the small example to tempt IPOPT solver. It works!

My suspicious is that the Matlab compiler failed to automatically detect the IPOPT items required for Matpower 7.0 which leads to the lack of IPOPT in the compiled program.

rdzman, 2019-06-21:
👍  Glad you got it working.

closed 2019-06-21 by rdzman

sususweet, 2021-07-06:
@shiftlin89 I also ran into this problem recently with MATPOWER 7.1 and MATLAB R2019b. I wonder how "integrate your example into the Matpower 7.0 , using the small example to tempt IPOPT solver" means. Can you give an example? Thanks.

shiftlin0818, 2021-07-06:
> @shiftlin89 I also ran into this problem recently with MATPOWER 7.1 and MATLAB R2019b. I wonder how "integrate your example into the Matpower 7.0 , using the small example to tempt IPOPT solver" means. Can you give an example? Thanks.

In your main function, add this code:
"
% !!!!! very important for release version
Lz_tempt_ipopt_into_compiler();
"

where function Lz_tempt_ipopt_into_compiler() is:
"
function Lz_tempt_ipopt_into_compiler()
%  This is very important to release , this progarm is used to link IPOPT 
%  when compile it
% 2019-06-21

fprintf('exist returns: %d\n', exist('ipopt', 'file'));
str = evalc('qps_ipopt([],[1; 1],[1 1],[2],[2],[1; 1],[1; 1],[1; 1],struct(''verbose'', 2))');
if ~isempty(str)
    fprintf('>>success to tempt IPOPT......\n');
end
"

This works for my program, good luck!

sususweet, 2021-07-07:
> In your main function, add this code:
> "
> % !!!!! very important for release version
> Lz_tempt_ipopt_into_compiler();
> "
>
> where function Lz_tempt_ipopt_into_compiler() is:
> "
> function [...]

Thanks! Another question is which method is used to compile the matlab .m files, mcc command line or MATLAB application compiler GUI? For the latter one, IPOPT can be successfully integrated.

shiftlin0818, 2021-07-07:
> Thanks! Another question is which method is used to compile the matlab .m files, mcc command line or MATLAB application compiler GUI? For the latter one, IPOPT can be successfully integrated.

I use mcc -m "*.m" file

sususweet, 2021-07-07:
> I use mcc -m "*.m" file

Thanks, I will have a try!