MATPOWER/matpower #73: Load Modeling for Radial load flow

akhilesh-barnwal, 2019-07-16:
I want to get the result for different type of voltage dependent load model. How can I code all these using radial power flow in MATPOWER?

akhilesh-barnwal, 2019-07-20:
In matpower, there is a function makeSdzip, which forms a vector of nominal value for each component of the ZIP load. Can anyone tell me how to use it?

rdzman, 2019-07-26:
Have a look at `makeSbus()` which is called inside `newtonpf()` via a function handle created in `runpf()`. It essentially calls `makeSdzip()` to get the ZIP proportions (as defined in the `exp.sys_wide_zip_loads.pw` option) ...

```
Sd = makeSdzip(baseMVA, bus, mpopt);
```

... then computes the complex power based on the voltage magnitude ...

```
Sbusd = Sd.p + Sd.i .* Vm + Sd.z .* Vm.^2;
```

Hope this helps.

akhilesh-barnwal, 2019-07-31:
Thank you, sir, You make me clear the issue.

akhilesh-barnwal, 2019-07-31:
and one more thing How could I go further to select some pre-decided particular buses (not all) to selected as having ZIP load? Please light on.

rdzman, 2019-07-31:
Unfortunately, the ZIP load option in the current implementation only has a single set of parameters that are applied to all loads. So you would have to supply your own per-load input data somehow to specify the ZIP proportions for each load.

closed 2019-08-16 by rdzman

curiousredfox, 2022-04-06:
Dear akhilesh-barnwal,
I'm also doing the same purpose like yours. I want to calculate the load flow with different ZIP load parameter at each load. I've been struggling for months not know how to modify the code "runpf". 
Could you help me with this? 
Please leave email so we can discuss further.
Thank you very much