MATPOWER/matpower #74: something wrong with “test_matpower”

JaneCheung, 2019-07-29:
whiel run the test_matpower:
...
t_opf_softlims..........**Error using loadcase (line 196)
loadcase: syntax error or undefined data matrix(ices) in the file
Error using feval
Undefined function or variable 'softlims_savecase_test_87757393'.

Error in t_opf_softlims (line 382)
mpc1 = loadcase(fn);

Error in t_run_tests (line 61)
    feval( test_names{k}, ~verbose );

Error in test_matpower (line 145)
all_ok = t_run_tests( tests, verbose );

Error in install_matpower (line 389)
            test_matpower;**
 
[image]
BUT I COULD FIND that test case!
How to deal with that problem!

rdzman, 2019-07-29:
This is very strange. Is it possible that the permissions on the `softlims_savecase_test_87757393.m` file do not include read permissions? That file is created by the test script using `savecase()` then immediately loaded using `loadcase()` (then deleted), so I'm not sure what could cause this error.

Can you just try running `t_opf_softlims` by itself to see if the error is repeatable?

And, btw, feel free to delete any of these `softlims_savecase_test_????????.m` files that are left behind. They are supposed to be automatically created and deleted on-the-fly for each test run.

rdzman, 2019-08-16:
Feel free to re-open this issue if you haven't been able to resolve it.

closed 2019-08-16 by rdzman