MATPOWER/matpower #80: Matpower 7 installation error on Matlab 2019a

nhelaly102, 2019-08-21:
I am facing an error when I try to install Matpower 7 on Matlab 2019a 

the error starts at:
t_opf_fmincon...........Undefined function or variable 'getIpOptions'.

although when I tried installing same package on an older version of Matlab, it was installed successfully.
[image]

rdzman, 2019-08-21:
This looks like a MATLAB installation issue. The function it says it can't find is `getIpOptions` which is part of the Optimization Toolbox and is called internally by `fmincon`. I'd try to reinstall MATLAB and see if that resolves the issue.

FYI, the tests all run fine on MATLAB R2019a on my machine.

nhelaly102, 2019-08-21:
I tried reinstalling the Matlab again including some of the missing toolboxes, and the problem was resolved.
Thanks so much.

closed 2019-08-21 by nhelaly102