MATPOWER/matpower #81: AC power flow with multiple refrence buses

gridxt, 2019-09-03:
As far as I know, MATPOWER ACPF algorithm cannot handle cases with multiple reference buses. However, there are 3 reference buses defined in the case_SyntheticUSA.m file. How come the 'runpf' command for this case still works? I suspect these buses are ignored...

rdzman, 2019-09-03:
If there are islands in the network, such as in `case_SyntheticUSA.m`, each island needs to have its own reference bus.

On the other hand, multiple reference buses in a fully connected network will cause problems.

gridxt, 2019-09-03:
Thanks Ray! Quick question: where in the code do you check for islands? I cannot find it...

rdzman, 2019-09-04:
The power flow and OPF do not check explicitly, it's just that there will likely be numerical issues if you do not have exactly one reference bus per island.

You can use `case_info()` or `find_islands()` to explicitly check for islands.

closed 2019-09-04 by rdzman