MATPOWER/matpower #82: Newton updates in cartesian voltage PF

rdzman, 2019-09-17:
@BaljaaSS, I was looking at the various power flow formulations again and would like to revisit the question I brought up [here][1] about Newton updates in the cartesian coordinate Newton power flow (in `newtonpf_S_cart()`). I realize I never responded to [your explanation][2].

I'm not sure I quite follow everything you said. If I understand correctly, equation (2) from your paper (doi: [10.1016/j.cam.2019.04.007][3]) was used to modify the Jacobian columns for ∆Vi for PV buses, to eliminate the corresponding ∆Vr column. So rather than solving for a ∆Vr and ∆Vi that keep Vm constant, we approximate that by assuming they keep V on the line tangent to the constant Vm circle at the current point. Then we use the linear approximation for the relationship between ∆Va and ∆Vr, ∆Vi from (3) to update only Va, leaving Vm constant. So your method for updating V at PV buses does make sense to me, as long as the real part of V never approaches zero, which it seems could cause numerical robustness issues. (Btw, are you concerned about this? If not, why not?)

However, I don't see any reason to use the approximations for ∆Vm and ∆Va from (2) and (3) to update V at PQ buses, as opposed to updating the real and imaginary parts of V directly. The Jacobian for these buses has not been modified at all using (2) or (3), so why should we do this extra approximation?

So, why not use the first alternative I proposed [here][1]? Are you sure this approach is less robust than the current one on large systems? I still can't see why it should be.

And, unfortunately, I have no familiarity with pre/post-smoothing in multigrid, so I'm afraid that doesn't help me 😜. 

[1]: https://github.com/MATPOWER/matpower/pull/43#issuecomment-465791438
[2]: https://github.com/MATPOWER/matpower/pull/43#issuecomment-465927259
[3]: https://doi.org/10.1016/j.cam.2019.04.007

BaljaaSS, 2019-10-04:
Dear Ray,

When I developed the various PF formulations, I have done the sensitivity analysis on different ways to update V as:

    Current Matpower implementation
    Directly updating real and imaginary parts as:         
            if npv
                Vi(pv) = Vi(pv) + dx(j1:j2);
                Vr(pv) = (abs(V(pv)).^2 - Vi(pv).^2).^(0.5);
            end   
            if npq
                Vi(pq) = Vi(pq) + dx(j3:j4);
                Vr(pq) = Vr(pq) + dx(j5:j6); 
            end
            V = complex(Vr,Vi);

As far as I remember, the current implementation in Matpower was the most robust one. For some cases, the second option gave me more iterations than the first one.

I think you can still update V using your proposed way. It looks faster and should also work without any problem. But I am not sure if that is the most optimal way.

For updating V at PV buses, we should still update it because we are building Jacobian matrix using updated V. In the current implementation, I am updating all V in polar coordinates, where V= Vm .* exp(1j * Va). Since Va(pv) is changed, V should be updated accordingly right?

I hope I explained it better than last time.

Regards,

Baljaa

rdzman, 2019-10-07:
Ok, so is it fair to say that the choice was based primarily on empirical testing, rather than on some compelling theoretical reason? In that case, I may do some of my own comparisons before making a final determination.

Also, it looks like your second option above is a slightly simpler form of the [3rd option I mentioned](https://github.com/MATPOWER/matpower/pull/43#issuecomment-465791438). (Except I think you have `j1:j2` and `j3:j4` swapped, probably based on an earlier version of the code.)

However, I did want to hear your thoughts (or those of anyone else listening) on the other question I raised about numerical issues that could appear as the real part of V approaches zero. We end up with a Jacobian entry that approaches infinity. This will yield a tiny value of ∆Vi that gets divided by a near-zero value to yield ∆Va.

If we know that Vr is close to zero (i.e. Va close to 90 deg), it would be much more numerically stable to eliminate ∆Vi from the update equation rather than ∆Vr.

rdzman, 2019-10-08:
The more I look at the details of this, the less comfortable I am calling this a standard Newton power flow in Cartesian coordinates. It really is a hybrid. We may be computing a Newton update in Cartesian coordinates, but the update we actually apply to the voltages is something quite different from this standard Newton update ... something that I suspect is quite a bit closer to the Newton update in polar coordinates.

For each PV bus, we have said that ∆Vr (and ∆Va, for that matter) are proportional to ∆Vi, so we eliminate ∆Vr and combine the columns of the Jacobian to compute only ∆Vi with the effect of the corresponding ∆Vr included. Then we scale ∆Vi by the appropriate proportion to get ∆Va.

We could just as easily combine the columns of the Jacobian for ∆Vr and ∆Vi to compute ∆Va directly for PV buses (and this is more robust because it avoids the issue mentioned above related to dividing by a potentially small Vr).
```
dSbus_dVi(:, pv) = ...
    dSbus_dVi(:, pv) * sparse(1:npv, 1:npv, real(V(pv)), npv, npv) - ...
    dSbus_dVr(:, pv) * sparse(1:npv, 1:npv, imag(V(pv)), npv, npv);
```
... and ...
```
if npv
    Va(pv) = Va(pv) + dx(j3:j4);
end
if npq
    Vm(pq) = Vm(pq) + (real(V(pq))./Vm(pq)) .* dx(j1:j2) ...
                    + (imag(V(pq))./Vm(pq)) .* dx(j5:j6);
    Va(pq) = Va(pq) + (real(V(pq))./(Vm(pq).^2)) .* dx(j5:j6) ...
                    - (imag(V(pq))./(Vm(pq).^2)) .* dx(j1:j2);
end
V = Vm .* exp(1j * Va);
```
But, however you look at it, in each case we are essentially computing a polar update of Va for PV buses, equivalent to using the appropriate columns directly from the polar Jacobian.

And the only reason I can see for using the polar update for the PQ buses, as we currently do, is because it is more robust. But again, it is a polar update computed from the values of a Cartesian Newton update.

All that to say, while our current "Cartesian" Newton power flow algorithms may work pretty well, they really are hybrid methods, not standard Newton in Cartesian coordinates.

Thoughts?

On another note, have you implemented the Cartesian version with the explicit Vm^2 constraint for PV buses? And did you observe it to be less reliable and worse convergence as concluded in ref [33] from your paper? It seems that is likely to be closer to what people would expect for a standard Newton power flow in Cartesian coordinates, don't you think?

rdzman, 2019-10-11:
I just implemented the this more standard cartesian Newton method with explicit voltage constraints. You can find it in the [`newtonpf-standard-cartesian`](https://github.com/MATPOWER/matpower/tree/newtonpf-standard-cartesian) branch. Comments are appreciated. I think it's actually *more* robust than the current method, but definitely slower, due to the larger number of variables and constraints.

BaljaaSS, 2019-10-13:
I agree that it would be more numerically stable to eliminate ∆Vi from the update equation rather than ∆Vr, as you explained. I was comparing many variants of the Newton power flow method with numerous ways to update V. I think I have not eliminated ∆Vi and compared it to others because removing ∆Vr was working well on all Matpower test cases. It is easy to change that on code. 

I would say **Current** mismatch is better between formulations, and **Polar** is better between coordinates. I did not call "Cartesian" Newton power flow algorithms **hybrid** because the main matrix equation is computed in Cartesian coordinates even though it is altered w.r.t **PV** buses. Also, all derivates are calculated in Cartesian coordinates.  As you said, since we are updating V in polar coordinates for Cartesian versions, we can call it Hybrid. As you already know that, updating V in polar coordinates for Cartesian versions is more robust than directly using Cartesian coordinates.

When I compared all variants, [33] using V(pv) .* conj(V(pv)) - Vmpv.^2  did perform poorly for large test cases. I saw your implementations on Cartesian Newton variants. I agree that it is more robust now due to extra variables and constraints.  It seems like you made [33] more robust. So I do not know if the new Cartesian variant is better than our hybrid one. 

Do you plan to make a comparison study?

rdzman, 2019-10-14:
Thanks for the comments. Just to clarify, I was not suggesting that eliminating ∆Vi is always more robust than eliminating ∆Vr. It depends on whether one of them is close to zero. The obvious way to ensure we have a numerically stable update that maintains the fixed proportion between ∆Vr and ∆Vi when either is close to zero is to compute the step length, not in the Vr or the Vi direction, but in the direction perpendicular to (Vr, Vi), that is the (Vi, -Vr) direction (see [this comment](https://github.com/MATPOWER/matpower/issues/82#issuecomment-539730853) above). In other words we compute ∆Y = Vr * ∆Vi - Vi ∆Vr. It turns out that this is identical to computing ∆Va directly. That is, the Jacobian column obtained by combining the columns for Vr and Vi in this way, **is identical** to the column relative to Va taken from the polar Jacobian.

In my testing, as expected, this approach is more robust than either just eliminating ∆Vr or eliminating ∆Vi, and results in a relatively robust and efficient hybrid method.

I don't really have time to do a rigorous comparison study at the moment. My current thought is to modify the MATPOWER 7 "cartesian" implementations with the above modification and rename it as a "hybrid" method and add the slower standard cartesian Newton method with the explicit PV bus voltage equations as the "cartesian" methods. What do you think of that idea?

The only complication is, given the current set of user options, there is not an obvious way to allow the user to specify this "hybrid" method. Setting `pf.v_cartesian` to something other than 0 or 1 would work, but isn't as intuitive as I'd like.

rdzman, 2019-10-16:
Here are some test results generated using the attached code (for future reference). Requires first creating `randsamples.mat` with an `rm` variable equal to `rand(140,000, 40)`, and running `newtonpf_testing` once with `reps` = 1 and the `Vm = ...; Va = ...; save(...)` lines uncommented. Then you re-comment those lines, set reps to the desired value, and uncomment the line for the value of `pf.alg` you want to test.

- [newtonpf_testing_results.txt](https://github.com/MATPOWER/matpower/files/3736183/newtonpf_testing_results.txt)
- [newtonpf_testing.m.txt](https://github.com/MATPOWER/matpower/files/3736189/newtonpf_testing.m.txt)

closed 2019-10-16 by rdzman

BaljaaSS, 2019-10-17:
You added new options for PF algorithm choice as NR-IH and NR-SH. Choosing the variant is clear for those people who know the difference between NR-IC and NR-IH, for example. Therefore, you need some report/technical note/paper to clarify these different approaches addition to mine. I also checked the test results. For me, it looks like Current mismatch formulation is the best overall and NR-IP is the fastest.
Furthermore, NR-IC converges after only two iterations for most of the  Distribution test cases. I think it is very interesting to study which method is more suitable for what type of network. What was your general conclusion based on test results?  I am quite interested in that.

rdzman, 2019-10-18:
I did update sections 4.1.1 and 4.1.2 in the manual to describe the hybrid method, which is just a slight tweak from what you implemented (to make it more numerically robust when Vr is small). The manual mentions that it's based on your paper. The description is minimal, but I hope sufficient for someone familiar with your paper.

Your conclusions from the results sound reasonable, but honestly I didn't really spend much time trying to interpret the results. My priority was on getting accurate and efficient implementations of the different variants that reflect the expectations of users given the names/descriptions.

commit df89b9e6 (2019-10-16): Redesign cartesian Newton power flow to use standard Newton method.

Redesign implementation of Newton power flow for cartesian voltage representations to use standard Newton method with explicit voltage magnitude constraints at PV buses. This method is slower (larger number of equations and unknowns) but appears to be more reliable in some cases than the original implementation, which was
actually a hybrid approach by Sereeter that used a modified Newton method to compute polar voltage updates using a cartesian Jacobian.

Closes #82.

Ref: https://github.com/MATPOWER/matpower/issues/82
Ref: h [... cut]