MATPOWER/matpower #85: error in test_matpower in matlab 2019b

downbydawn, 2019-11-14:
I have installed matpower 7.0 in Matlab 2019b, when I run test_matpower I get:

> Unrecognized function or variable 'getIpOptions'.

which causes:

> Error in fmincon (line 822)
    options = getIpOptions(options,sizes.nVar,mEq,flags.constr,defaultopt,10,0.01);

How can I solve this?

rdzman, 2019-11-14:
It seems like there is an issue with your MATLAB installation, specifically with the Optimization Toolbox. It can't find the getIpOptions function, which is a part of the Optimization Toolbox. On my machine, `which getIpOptions` returns the following ...
```
>> which getIpOptions
/Applications/-math/MATLAB_R2019b.app/toolbox/optim/optim/getIpOptions.m
```

Based on the error message, I'm guessing on your machine it'll simply say ...
```
>> which getIpOptions
'getIpOptions' not found.
```

If so, I'd suggest attempting to re-install MATLAB.

downbydawn, 2019-11-18:
installing the optimization toolbox worked, thank you :)

closed 2019-11-18 by downbydawn

Kali-Nath, 2020-04-10:
Hi,
I had encountered the same error but got it resolved after I installed the toolbox, thank you very much.