MATPOWER/matpower #88: Matpower 7 installation error on Matlab 2019a Unrecognized function or variable 'getIpOptions'.

aliamin12, 2020-01-29:
[image]

Hi there, I keep getting this error whilst trying to install Matpower 7.0. I have reinstalled Matlab as this was advised to a previous user, and that worked for them. However, after reinstalling I still have the same error. 

Please could you look into this for me. 

Thanks

rdzman, 2020-01-29:
It looks like there is something unexpected about your Optimization Toolbox installation. Can you post the results of the following commands in MATLAB?

```
ver
exist('fmincon')
which fmincon
```

aliamin12, 2020-01-29:
[image]

Please see attached.

rdzman, 2020-01-29:
It looks like the `fmincon.m` file is present, which is how MATPOWER attempts to detect the presence of the Optimization Toolbox. But, for some reason it is not recognized by MATLAB. Do you have a license for it?

rdzman, 2020-01-29:
You should be able to test for that with ...

```
license('test', 'optimization_toolbox')
```

rdzman, 2020-01-29:
I'm assuming that will return a true value (since MATPOWER also checks that). So it seems to me there's something wrong with the MATLAB installation, specifically regarding the Optimization Toolbox. What happens if you try to call `fmincon()`?

aliamin12, 2020-01-29:
[image]

Yes, I believe it is, as my friends have managed to download just fine and they use the same institutional license.

aliamin12, 2020-01-29:
[image]

rdzman, 2020-01-29:
Can you try the following (an example from `help fmincon`) ...
```
X = fmincon(@(x) 3*sin(x(1))+exp(x(2)),[1;1],[],[],[],[],[0 0])
```

rdzman, 2020-02-07:
I'm pretty sure this is a MATLAB/Optimization Toolbox installation issue, but please feel free to re-open this issue if it turns out to be related to MATPOWER.

closed 2020-02-07 by rdzman