MATPOWER/matpower #90: CPLEX 12.10 causing warnings about deprecated toolbox parameters

rdzman, 2020-02-07:
With CPLEX 12.10, calling CPLEX via the toolbox interface (as opposed to the class interface), as MATPOWER does, results in lots of warnings like the following. Thanks to Rizwan Ali for bringing this to our attention.

```
Matlab Toolbox parameter "Display" found.
Warning: Using Matlab Toolbox parameters in the CPLEX Matlab API is deprecated.
They will be treated as unknown parameters in the future. 
> In handleparams
  In cplexqcp
  In cplexlp
  In qps_cplex (line 249)
  In qps_matpower (line 231)
  In t_qps_matpower (line 85) 
```

It doesn't appear that these warnings can be turned off using `warning()`.

It also isn't clear how, or even whether, the progress display can be controlled via the toolbox API at all anymore as of CPLEX 12.10.

As a workaround, the warnings can be silenced (at the cost of completely disabling all CPLEX progress output), by commenting out [lines 125-131 in `cplex_options.m`](https://github.com/MATPOWER/matpower/blob/6fdd814610dabca1ce8464dc9675ae02bbcd7c7c/lib/cplex_options.m#L125-L131) and [lines 215-221 in `qps_cplex.m`](https://github.com/MATPOWER/matpower/blob/6fdd814610dabca1ce8464dc9675ae02bbcd7c7c/lib/qps_cplex.m#L215-L221), and [lines 231-237 in `miqps_cplex.m`](https://github.com/MATPOWER/matpower/blob/6fdd814610dabca1ce8464dc9675ae02bbcd7c7c/lib/miqps_cplex.m#L231-L237).

However, given that the toolbox interface has a lot of overhead anyway (calling `cplexoptimset('cplex')` in `cplex_options()` is a particular bottleneck for small problems), it seems the proper solution to this issue will be to switch the `qps_cplex()` and `miqps_cplex()` implementations to use the class API instead of the toolbox API.

rdzman, 2020-02-26:
Well, it turns out I was wrong about all of this.

1. The toolbox API still works just fine with CPLEX 12.10. You just have to use `display` instead of `Display` to control the output.
2. The calls to `cplexoptimset()` are not necessary for using the toolbox API and they really were the only bottleneck. I discovered this after implementing a prototype version of `qps_cplex()` that uses the class API, and found that the toolbox API version was slightly faster than the class API version once I got rid of the calls to `cplexoptimset()`.

closed 2020-02-26 by rdzman

parole718, 2022-11-12:
Hi, I am curious about how to use display instead of Display. What alterations should I make? As far as I can see, the problem is with the CPLEX. Matlab has been notifying such warnings all the time and it's just really annoying because I am unable to turn off the alarm.

rdzman, 2022-11-14:
What version of MATPOWER are you using? (Type `mpver` to display the version number.)

This issue was fixed back in 2020, before the release of MATPOWER 7.1. So the only thing you should need to do is make sure you have MATPOWER 7.1 or later.

parole718, 2022-11-15:
Hey Ray, 
I am rather surprised and delighted when I heard from you last night. 
First of all, I sincerely appreciate your reply. I want to introduce myself a little bit. I am an undergraduate student and majors in management. I intend to use the CPLEX software to solve the MIP problem and finish a thesis. I only have shallow understanding of the Matlab software.
It turns out that I use the Matlab software instead of matpower, which is a software derived from the Matlab. As a result, when I typed the mpver command in the command window, the Matlab software raised an error signal.
In your reply, you mentioned that the bug was fixed but the warning signal constantly appeared whenever I run the CPLEX. As far as I am concerned, the warning actually has no effect on the final result, but I have zero tolerance to the bug. I have scoured the Internet for a solution, but I barely found any useful information. I traced the cause all the way to the source file, and I found the two lines of code which gave rise to the warning. The problem is that if I annotated the lines, the whole code would cease to function normally.
I attach some photos to the mail to better describe the dilemma. Your reply is more than welcome.
Yours sincerely,
      Tom

rdzman, 2022-11-15:
You may want to check out [MP-Opt-Model](https://github.com/MATPOWER/mp-opt-model/), which provides a unified interface for multiple solvers (including CPLEX). It is was developed as the optimization model interface for MATPOWER, but it is applicable to general optimization problems.

It handles silencing the CPLEX warnings you are having trouble with. If you do choose to use it for your problem, it will allow you to switch solvers (e.g. between CPLEX, GLPK, Gurobi, MOSEK or MATLAB's own Optimization Toolbox solvers) by changing a single option. It also has some nice facilities for building your model, if that would help.

If, on the other hand, you do need to interface directly to CPLEX, maybe this will help. Here is the [commit](https://github.com/MATPOWER/mp-opt-model/commit/55287210a294faf7e0bc9fcebfdbea1f7f9793e7) where I fixed the problem above. The field name in the options struct you pass to `cplexlp()`, `cplexqp()`, `cplexmilp()`, or `cplexmiqp()` needs to be changed from `Display` to `display`.

parole718, 2022-11-16:
Hey Ray,
Thanks for your reply!
I was assigned to use the CPLEX solver, otherwise I could have adopted Matlab original toolbox, and that's to say I have to interface directly to CPLEX.
I have noticed the commit, which is how to fix the problem in MATPOWER. However, there are some major DIFFERENCES between MATLAB and MATPOWER, and there isn't a cplex_options.m document in my MATLAB software. Consequently, I cannot lowercase the capital letter 'D'. I have also searched through all the available .m documents in the CPLEX file. Unfortunately, I still cannot find the cause to the warning and most of the documents comprise entirely of annotations, which is quite distressing.
To successfully interface directly to the CPLEX, I also add the YALMIP-master file to the MATLAB toolbox. I attach the relevant zipped files (the CPLEX file and the YALMIP-master) to better explain my problem.
Your response is more than welcome. 
p.s. I think I should ignore the problem. After all, it is a minor warning that a coding expression is out-dated
Yours sincerely,
Tom

 

 
 
 

从QQ邮箱发来的超大附件

cplex.zip (74.27M, 2022年12月16日 13:43 到期)进入下载页面：http://mail.qq.com/cgi-bin/ftnExs_download?t=exs_ftn_download&k=5f3335348591389ee2d2756e1735514a0756050c045151034c0453520918575d530018560956574850040306550c5a56530a060c313c6306115f504c1f4f0a15610e&code=a35415ce

rdzman, 2022-11-16:
Ok, so you are interfacing to CPLEX through YALMIP, rather than calling one of the CPLEX functions directly. Yeah, unfortunately, I'm not familiar with how to set CPLEX options when using YALMIP, but I suggest you take your question to the [YALMIP discussion forum](https://github.com/yalmip/YALMIP/discussions).