MATPOWER/matpower #93: Data question

MrWangRobot, 2020-03-29:
Excuse me, why can't I find loss data in results struct? As shown in the figure:
[image]

[image]
Where can I find  it?

rdzman, 2020-03-30:
It is computed from the data in the `results` struct, but not stored in `results`. You can use the [`get_losses()`](https://matpower.org/docs/ref/matpower7.0/lib/get_losses.html) function to compute the components of the branch losses.

closed 2020-03-30 by rdzman

MrWangRobot, 2020-03-31:
I got it.  thank you for your  help.