MATPOWER/matpower #94: IEEE 118 solution difference with comercial tools

SanPen, 2020-04-02:
Hi, 

I am doing a computer program to run power flows and I have done it to the image of Matpower, following the same matrix formulation. In general my program provides the same results as MatPower.

One of the tasks is to compare the resulting tool with comercial alternatives. In my case I have used PSS/e and Power Factory.

I have observed two issues that attain to MatPower as well:

- The first is that MatPower IEEE118 bus case, follows the IEEE118 bus RAW file to the letter. This is usually good, but it happens that the IEEE118 RAW file contains 6 repeated branches that comercial tools omit. In the comparison file, you will find a comparison between my tool and the commercial ones with the repeated branches omitted. Also find attached Matpower's branch results.

- The second is more subtle. Even after removing the duplicated branches, the results of the commercial tools agree perfectly, the tool I have made and Matpower agree perfectly, but both groups disagree. Meaning that Matpower or my tool do not offer the same solution as PSS/e or Power Factory.

The first Issue doesn't worry me as much as the second. 

Could the results difference be due to a modelling difference?
(the commercial software runs have been done with no controls activated)

Do you have any clue of why could this be?

For your information, IEEE14 and IEEE30 match perfectly.

Thank,
Santiago

[Matpower_IEEE118.txt](https://github.com/MATPOWER/matpower/files/4420600/Matpower_IEEE118.txt)
[ieee118_power_factory.txt](https://github.com/MATPOWER/matpower/files/4420665/ieee118_power_factory.txt)
[IEEE_118_comparison.xlsx](https://github.com/MATPOWER/matpower/files/4420582/IEEE_118_comparison.xlsx)
[IEEE118.xlsx](https://github.com/MATPOWER/matpower/files/4420583/IEEE118.xlsx)
[IEEE 118 Bus.raw.txt](https://github.com/MATPOWER/matpower/files/4420587/IEEE.118.Bus.raw.txt)

SanPen, 2020-04-02:
Well,  Fixed the second issue was the slack; The .raw file defines bus 80 as the area slack, whereas in the buses section the bu 69 is the slack. Correcting this the results are exact.

Find attached a corrected RAW file.

[IEEE 118 Bus v2.raw.txt](https://github.com/MATPOWER/matpower/files/4422425/IEEE.118.Bus.v2.raw.txt)

rdzman, 2020-04-02:
Glad to hear you found it and were able to get the results to match. Nice work.

closed 2020-04-02 by rdzman