MATPOWER/matpower #97: Warning: Matrix is singular to working precision in MATPOWER

allrashdi10, 2020-04-11:
Hello,
I'm working in my FYP which needs to make a study using the MATPOWER.
I was enter all Power Flow data (Generation data, Bus data and Branch data) correctly.
when I run this code
```
define_constants;
r=runpf(OETC_PF);
```
where OETC_PF is my Power Flow data.
and I get this message :

```
MATPOWER Version 7.0, 20-Jun-2019 -- AC Power Flow (Newton)
Warning: Matrix is singular to working precision. 
> In mplinsolve (line 75)
  In newtonpf (line 110)
  In runpf (line 260)
  In Untitled (line 2)
 
Warning: Matrix is singular to working precision. 
> In mplinsolve (line 75)
  In newtonpf (line 110)
  In runpf (line 260)
  In Untitled (line 2)
 
Warning: Matrix is singular to working precision. 
> In mplinsolve (line 75)
  In newtonpf (line 110)
  In runpf (line 260)
  In Untitled (line 2)
 
Warning: Matrix is singular to working precision. 
> In mplinsolve (line 75)
  In newtonpf (line 110)
  In runpf (line 260)
  In Untitled (line 2)
 
Warning: Matrix is singular to working precision. 
> In mplinsolve (line 75)
  In newtonpf (line 110)
  In runpf (line 260)
  In Untitled (line 2)
 
Warning: Matrix is singular to working precision. 
> In mplinsolve (line 75)
  In newtonpf (line 110)
  In runpf (line 260)
  In Untitled (line 2)
 
Warning: Matrix is singular to working precision. 
> In mplinsolve (line 75)
  In newtonpf (line 110)
  In runpf (line 260)
  In Untitled (line 2)
 
Warning: Matrix is singular to working precision. 
> In mplinsolve (line 75)
  In newtonpf (line 110)
  In runpf (line 260)
  In Untitled (line 2)
 
Warning: Matrix is singular to working precision. 
> In mplinsolve (line 75)
  In newtonpf (line 110)
  In runpf (line 260)
  In Untitled (line 2)
 
Warning: Matrix is singular to working precision. 
> In mplinsolve (line 75)
  In newtonpf (line 110)
  In runpf (line 260)
  In Untitled (line 2)
 
Newton's method power flow (power balance, polar) did not converge in 10 iterations.

```

How can I slove it?

rdzman, 2020-04-11:
It seems likely that there is some problem with the case file, possibly a disconnected network or something. I would start looking at the output of `case_info('OETC_PF')` to see if it shows anything unexpected.

allrashdi10, 2020-04-12:
This is the output I get from `case_info`
```

>> Untitled
Checking connectivity ... 4 connected groups, 3 isolated buses
Elapsed time is 0.080425 seconds.
================================================================================
                        Full       Island      Island      Island      Island   
                       System         1           2           3           4     
Number of:           ----------  ----------  ----------  ----------  ---------- 
  buses                   126          84          26          11           2   
  loads                    86          72           8           5           -   
    on                     86          72           8           5           -   
    off                     -           -           -           -           -   
    fixed                  86          72           8           5           -   
    dispatchable            -           -           -           -           -   
      on                    -           -           -           -           -   
      off                   -           -           -           -           -   
  generators               10           1           6           2           1   
    on                     10           1           6           2           1   
    off                     -           -           -           -           -   
  shunt elements           74          65           4           5           -   
  branches                135          96          28          10           1   
    on                    135          96          28          10           1   
    off                     -           -           -           -           -   
    ties (off)              -           -           -           -           -   

Load                
  active (MW)       
    dispatched           7605.2      6863.6       424.2       223.5         -   
      fixed              7605.2      6863.6       424.2       223.5         -   
      dispatchable          -           -           -           -           -   
    curtailed               -           -           -           -           -   
    nominal              7605.2      6863.6       424.2       223.5         -   
      on                 7605.2      6863.6       424.2       223.5         -   
      off                   -           -           -           -           -   
      fixed              7605.2      6863.6       424.2       223.5         -   
      dispatchable          -           -           -           -           -   
        on                  -           -           -           -           -   
        off                 -           -           -           -           -   
  reactive (MVAr)   
    dispatched           2698.6      2455.4       138.9        73.5         -   
      fixed              2698.6      2455.4       138.9        73.5         -   
      dispatchable          -           -           -           -           -   
    curtailed               -           -           -           -           -   
    nominal              2698.6      2455.4       138.9        73.5         -   
      on                 2698.6      2455.4       138.9        73.5         -   
      off                   -           -           -           -           -   
      fixed              2698.6      2455.4       138.9        73.5         -   
      dispatchable          -           -           -           -           -   
        on                  -           -           -           -           -   
        off                 -           -           -           -           -   

Generation          
  active (MW)       
    dispatched           8230.1       241.9      4425.5      1711.6      1851.1 
    max capacity         8932.5       263.8      4811.5      1839.0      2018.2 
      on                 8932.5       263.8      4811.5      1839.0      2018.2 
      off                   -           -           -           -           -   
    min capacity         2679.8        79.1      1443.5       551.7       605.5 
      on                 2679.8        79.1      1443.5       551.7       605.5 
      off                   -           -           -           -           -   
  reactive (MVAr)   
    dispatched           3396.1        24.7      1700.9       837.8       832.7 
    max capacity         5359.5       158.3      2886.9      1103.4      1210.9 
      on                 5359.5       158.3      2886.9      1103.4      1210.9 
      off                   -           -           -           -           -   
    min capacity        -1607.9       -47.5      -866.1      -331.0      -363.3 
      on                -1607.9       -47.5      -866.1      -331.0      -363.3 
      off                   -           -           -           -           -   
[... 87 lines cut ...]

Shunt Injections    
    active (MW)             -   
    reactive (MVAr)         -   

Branch Losses       
    active (MW)             -   
    reactive (MVAr)         -   

DC line             
  export (MW)       
    dispatch                -   
    max capacity            -   
      on                    -   
      off                   -   
    min capacity            -   
      on                    -   
      off                   -   

Reference Buses     
  num of ref buses          0   
  ref bus numbers               
>> 

```

rdzman, 2020-04-13:
As I suspected. Your network is not fully connected. You have 4 islands, plus another 3 isolated buses. Make sure that all buses are connected via branches into a single network ... or at least that there are no isolated buses and you have defined a `REF` bus in each island and each island has adequate generator to supply its own load.

allrashdi10, 2020-04-15:
We found the 3 isolated buses and we put ref for them, and the result after that is :
```

>> Untitled33333

MATPOWER Version 7.0, 20-Jun-2019 -- AC Power Flow (Newton)

Newton's method power flow (power balance, polar) did not converge in 10 iterations.

>>>>>   Did Not converge (0.02 seconds)   <<<<<

```

How can we get rid of this?

-------------------------------

Also I would like to ask you about the purpose of islands. Because in our project we did not use islands or zones. I mean how can we know the way to separate the islands?

rdzman, 2020-04-15:
I think the power flow problem is still caused by a system that is not fully connected. Rather than changing isolated buses to type REF (which will only work if they have a generator that can meet all of their load), you probably want to add the missing branches between them and the rest of the system and between the disconnected islands. Normally you want all buses to be connected into a single network.

allrashdi10, 2020-05-14:
After solving the previous problem I proceed to find the OPF of the system but this problem is showing up.

```
>> dola

MATPOWER Version 7.0, 20-Jun-2019 -- AC Optimal Power Flow
  AC OPF formulation: polar voltages, power balance eqns
MATPOWER Interior Point Solver -- MIPS, Version 1.3.1, 20-Jun-2019
 (using built-in linear solver)Warning: Matrix is close to singular or badly scaled. Results may be inaccurate. RCOND =  1.714762e-16. 
> In mplinsolve (line 70)
  In mips (line 467)
  In mipsopf_solver (line 129)
  In opf_execute (line 86)
  In opf (line 232)
  In runopf (line 75)
  In dola (line 2)
 
Warning: Matrix is close to singular or badly scaled. Results may be inaccurate. RCOND =  7.953985e-18. 
> In mplinsolve (line 70)
  In mips (line 467)
  In mipsopf_solver (line 129)
  In opf_execute (line 86)
  In opf (line 232)
  In runopf (line 75)
  In dola (line 2)
 
Warning: Matrix is close to singular or badly scaled. Results may be inaccurate. RCOND =  3.907404e-19. 
> In mplinsolve (line 70)
  In mips (line 467)
  In mipsopf_solver (line 129)
  In opf_execute (line 86)
  In opf (line 232)
  In runopf (line 75)
  In dola (line 2)
 
Warning: Matrix is close to singular or badly scaled. Results may be inaccurate. RCOND =  3.104240e-21. 
> In mplinsolve (line 70)
  In mips (line 467)
  In mipsopf_solver (line 129)
  In opf_execute (line 86)
  In opf (line 232)
  In runopf (line 75)
  In dola (line 2)
 

Numerically Failed

Did not converge in 16 iterations.

>>>>>  Did NOT converge (1.16 seconds)  <<<<<

```
How to solve it ? :(

rdzman, 2020-05-14:
It's hard to say. These sorts of numerical failures can be for a variety of reasons.

The first one to check is for infeasibility of the problem. The easiest way to do that is to try relaxing all branch flow constraints completely by setting the `RATE_A` column to 0 ...
```
mpc.branch(:, RATE_A) = 0;
```
... and increasing the generator P and Q ranges sufficient to make sure they are not preventing feasibility. If that works, you can check which of the original limits are violated and begin tightening the constraints again until you identify which one(s) are making the problem infeasible.

If the problem appears to be feasible, you could try turning on the `mips.step_control` option. If that doesn't help, maybe try a different OPF solver. If you have the Optimization Toolbox, try setting `opf.ac.solver` to `FMINCON`. Or try installing IPOPT (e.g. via [OPTI Toolbox](https://www.inverseproblem.co.nz/OPTI/) if you are on MS Windows) and setting `opf.ac.solver` to `IPOPT`.

These are just some ideas ...

allrashdi10, 2020-05-17:
We try to put all branches equal to zero and then the system code is work! Now we want to know which one is the tightening line because we have 161 line. I mean is there a way to know directly which one is tightening ?

rdzman, 2020-05-18:
Try multiplying the `RATE_A` column by a scale factor instead to relax the limits. Once it is large enough to converge, you should be able to decrease it gradually and observe the shadow prices on the binding constraints `MU_SF` and `MU_ST` columns in `branch` to see which are the flows that bind first and hardest.

closed 2020-06-12 by rdzman