MATPOWER/matpower #103: Why are the generator costs for Polish grid two orders of magnitude higher than generator costs for French grid?

djinnome, 2020-06-09:
Hi folks,

Table B-4 on page 152 of the [Matpower Users manual](https://matpower.org/docs/manual.pdf) defines the costs in US Dollars. 

[image]

Since the generator cost function is a 1st order polynomial:

$$ f(p) = c_1p + c_0$$

 where $p$ is the power in MW, and $f$ is in \$/hour, so $c_1$ is in $\frac{\$}{MW\cdot hr}$.

I would expect that the generator costs should not vary too much among different cases.

However, the Polish power grid `case2383wp` has [1st order generator cost coefficients](https://github.com/MATPOWER/matpower/blob/master/data/case2383wp.m#L5657-L5693) of approximately $200 $/MW/hour

whereas the French power grid `case1888rte` has [1st order generator cost coefficients](https://github.com/MATPOWER/matpower/blob/master/data/case2383wp.m#L5657-L5693) of approximately $1 $/MW/hour.

And this difference of two orders of magnitude in 1st order generator cost coefficients is consistent between all French `rte` cases and Polish `wp` cases.  

Is it really the case that Polish power is really 100x more expensive than French power, or are the generator costs actually in different currencies (francs vs zlotys)?

rdzman, 2020-06-12:
I'm certain that Polish power is not really 100x more expensive than French power.

However, these cases were created by different teams, presumably for different purposes, so it is safe to say that directly comparing the costs is probably not appropriate.

I suggest contacting the creators of the cases directly to understand the meaning of the costs that have been included. Contact information should be listed in the help text at the top of the case file.

I notice for the Polish case many of the costs are zero. And for the French RTE case most costs are either 1 or 2, suggesting that they are not actual generation costs, but proxies used to produce some sort of rough dispatch ranking.

ccoffrin, 2020-06-12:
To add, I have a vague recollection that the RTE networks were configured with a line loss minimization objective rather than a fuel cost one.  This report should provide a detailed explanation, https://arxiv.org/abs/1603.01533

closed 2020-06-29 by rdzman