MATPOWER/matpower #105: Bug Report: runpf() doesnt converge when running on results obtained using runpf()

Vurgit, 2020-06-29:
https://github.com/Vurgit/case_IEEE123/blob/master/grid_IEEE123_complete.m
This case can converge, but if runpf() again on the results of the case, it doesnt converge! I have never seen this before and I think it might be a bug

rdzman, 2020-06-29:
I don't believe this is strictly speaking a bug.

It looks like this is a numerical issue specific to this case, caused by the extremely small reactance values (10^-8) on some of the lines. This results in a condition number greater than 10^9 for the Jacobian, making it very difficult to converge to the default tolerance of 10^-8. This is simply an example of a very ill-conditioned problem.

You can either solve the case with one of the radial methods (e.g. set `'pf.alg'` to `'PQSUM'`) or modify the case to make it less numerically problematic. For example, you could merge buses that are connected by extremely small impedances.

Vurgit, 2020-06-30:
Yes it is, the lines with extremely small reactances are closed breakers in the case, and their resistances are not given. They were supposed to be merged with other buses, but I still wanna keep them when doing OPF and other researches so I kept them and modelled them using extremely small reactances.

rdzman, 2020-06-30:
I suppose it would be useful for MATPOWER to provide the option to automatically merge nodes internally for the computation in such cases.

I would certainly consider including that feature if someone has the time and interest in implementing it.

closed 2020-07-16 by rdzman