MATPOWER/matpower #106: Need more messages from glpk

temcdrm, 2020-10-21:
Hello. I'm trying to solve a 24-period, deterministic unit commitment problem with matpower7.1 / octave 5.2.  It has 13 generators, 8 fixed loads and 8 dispatchable loads.  However, I can only get a solution when the dispatchable loads are "too big" by a factor of about 4. With realistic sizes it fails, even though opf solves.  I've tried this to get more information:

```
mpopt = mpoption('verbose', 3, 'out.all', 1, 'most.dc_model', 0, 'opf.dc.solver', 'GLPK');
mpopt = mpoption(mpopt, 'glpk.opts.msglev', 3);
```
Which produces:
```
- Building indexing structures.
- Building expected storage-tracking mechanism.
- Building constraint submatrices.
  - Building load balance constraints.
  - Building CCV constraints for piecewise-linear costs.
  - Building contingency reserve constraints.
  - Building ramping transitions and reserve constraints.
  - Building unit commitment constraints.
- Building cost structures.
- Assembling full set of constraints.
- Assembling full set of variable bounds.
- Assembling full set of costs.
- Calling MILP solver.

============================================================================

GLPK Integer Optimizer, v4.65
5022 rows, 5502 columns, 13269 non-zeros
312 integer variables, all of which are binary
Preprocessing...
**PROBLEM HAS NO PRIMAL FEASIBLE SOLUTION**

============================================================================
- MOST: MILP solver 'GLPK' failed with exit flag = -10
- Post-processing results.
- MOST: Done.
```
Is there any way to get more information, e.g., the equations passed to GLPK or the offending variables?

Thanks in advance for any advice given.

rdzman, 2020-10-22:
Unfortunately, I'm not aware of a way to get GLPK to give you more information about what is causing the infeasibility. If you are comfortable sharing the problem data, feel free to send the `most()` inputs (preferably attached here, but private e-mail is also an option) and I'll can try another solver such as Gurobi or CPLEX to see if they give more insight.

Another option is to try relaxing some constraints to see if you can get it to solve. For example, make sure all min up and down times are set to 0, or try setting all of the generator PMIN values to 0 and turn off the unit commitment by setting the `'most.uc.run'` option to 0. 

If one of these solves, the solution may give you some clues as to why the original is infeasible.

temcdrm, 2020-10-23:
[ercot8.zip](https://github.com/MATPOWER/matpower/files/5430772/ercot8.zip)
The files are attached; `test_damsolve` fails, but it succeeds if you change `scale = 4.0` in line 15 of test_damresp.m.  Please let me know if it works in a different solver.

rdzman, 2020-10-27:
After removing the `endfunction` lines (which is a syntax error in MATLAB), I was able to run the case with other solvers under MATLAB. Btw, `most.solver` is the correct option to use to control the solver for MOST, not `opf.dc.solver`. In any case, I got the same result, with CPLEX pointing to infeasible bounds on Pg(14).

After a closer look, I think I found the problem. I believe you may have misunderstood the way load scaling works. While you are only scaling the dispatchable loads by using `col = CT_LOAD_DIS_P` in `test_damresp.m`, the value you provide must be a **per-bus total** load, not just the value of the dispatchable load. So, for example, I believe given your data, it is setting the dispatchable load for bus 1 period 1 to -4788.43 in order to get the total to equal 2394.22 (=7182.65-4788.43). So, if you want the final *dispatchable* load to be 2394.22, as I assume you intended, you will have to add the value of the fixed load (7182.65) to the 2394.22 and specify the sum of the two in `test_damresp.m`.

And, in fact, adding the following two lines at the end of `test_damresp.m` does allow it to solve with scale = 1.0.
```
unresp = test_damunresp;
resp.values = resp.values + unresp.values;
```

temcdrm, 2020-10-28:
Thanks! Those two lines do what I want, i.e., independently control the fixed and dispatchable loads.  

Once we are finished testing, this model will be posted on GitHub to go with a paper that was published a few months ago in Applied Energy: [https://www.osti.gov/biblio/1646637](url)

Closing this issue because it's not possible to get more information from GLPK.

closed 2020-10-28 by temcdrm

rdzman, 2020-10-28:
Thanks for mentioning the paper. And please post a link (to MATPOWER-L, Power-Globe, etc.) to the test system when it is available. Working test systems are very useful and we don't really have that many for MOST yet.