MATPOWER/matpower #108: Add MUMPS as linear solver option to mplinsol

rwl, 2020-11-30:
[MUMPS](http://mumps.enseeiht.fr/) might be a good candidate for a new linear solver option in `mplinsol`. Benchmarking would be required, but its support for parallelism may make it a good alternative to PARDISO when solving large systems, but without the license restrictions.

It includes a [MATLAB/Octave interface](http://mumps.enseeiht.fr/doc/userguide_5.3.5.pdf#section.10). While this doesn't work with MPI-based distributed memory parallelism, it does support shared memory, multithreaded parallelism through OpenMP and multithreaded BLAS implementations.

rdzman, 2020-11-30:
Thanks for this suggestion.

Are you sure the MATLAB/Octave interface can support parallelism? Section 10 in the [User's Guide](http://mumps.enseeiht.fr/doc/userguide_5.3.5.pdf#section.10) states ...

_Only an interface to the sequential version of MUMPS is provided, thus only the parameters related to the sequential version of MUMPS are available._

In any case, as you say, it would be useful to have some benchmarks to get an idea of whether there are improvements over the built-in backslash operator, and if so, how much and under what circumstances.

rdzman, 2020-11-30:
Moved to MATPOWER/mips#2, since that's the master repo for `mplinsolve()`.

closed 2020-11-30 by rdzman