MATPOWER/matpower #109: CPF results struct only contains last event

rwl, 2020-12-04:
Table 5-1 of the MATPOWER Manual states that the `events` field is:

https://github.com/MATPOWER/matpower/blob/4ef0fc187a0df0799b701ee1d15d1a57f7a45beb/docs/src/MATPOWER-manual/MATPOWER-manual.tex#L1831

However, running this example from the manual:

```matlab
define_constants;
mpopt = mpoption('out.all', 0, 'verbose', 2);
mpopt = mpoption(mpopt, 'cpf.stop_at', 'FULL', 'cpf.step', 0.2);
mpcb = loadcase('t_case9_pfv2');
% load base case
mpct = mpcb;
% set up target case with
mpct.gen(:, [PG QG]) = mpcb.gen(:, [PG QG]) * 2.5; % increased generation
mpct.bus(:, [PD QD]) = mpcb.bus(:, [PD QD]) * 2.5; % and increased load
results = runcpf(mpcb, mpct, mpopt);
results.cpf.events
```

Output:
```
ans =

  scalar structure containing the fields:

    k =  16
    name = TARGET_LAM
    idx =  1
    msg = ZERO detected for TARGET_LAM event : CONTINUE
```

It appears that only the last event is returned.

abhyshr, 2020-12-04:
There is only one event for the above case - terminate on tracing the full continuation curve back to lam = 0. Hence CPF reports only one event. The `idx` field is the event number, which in this case is 1 since there is only one event.

If you run with reactive power llmits enforced then you'll see there are are two events. The first one is for reaching the reactive power limit of a generator and the second for tracing the full continuation curve.

```
>> mpopt.cpf.enforce_q_lims = 1;
>> results = runcpf(mpcb, mpct, mpopt);

MATPOWER Version 7.0b1, 31-Oct-2018 -- AC Continuation Power Flow
step   0  :                      lambda =  0.000,  4 Newton steps
step   1  : stepsize = 0.2       lambda =  0.181   2 corrector Newton steps
step   2  : stepsize = 0.2       lambda =  0.359   2 corrector Newton steps
step   3  : stepsize = 0.2       lambda =  0.530   2 corrector Newton steps
step   4  : stepsize = 0.2       lambda =  0.693   3 corrector Newton steps
step   5  : stepsize = 0.2       lambda =  0.839   3 corrector Newton steps
step   6  : stepsize = 0.2       lambda =  0.952   3 corrector Newton steps
step   7a : stepsize = 0.2       lambda =  0.988   3 corrector Newton steps ^ ROLLBACK
step   7  : stepsize = 0.00398   lambda =  0.953   1 corrector Newton steps
step   8  : stepsize = 0.000569  lambda =  0.954   1 corrector Newton steps
step   9  : stepsize = 8.31e-05  lambda =  0.954   1 corrector Newton steps
step  10  : stepsize = 0         lambda =  0.954   1 corrector Newton steps
step  11  : stepsize = 0.2       lambda =  0.968   3 corrector Newton steps
step  12  : stepsize = 0.2       lambda =  0.947   3 corrector Newton steps
step  13  : stepsize = 0.2       lambda =  0.899   2 corrector Newton steps
step  14  : stepsize = 0.2       lambda =  0.832   2 corrector Newton steps
step  15  : stepsize = 0.2       lambda =  0.747   2 corrector Newton steps
step  16  : stepsize = 0.2       lambda =  0.649   2 corrector Newton steps
step  17  : stepsize = 0.2       lambda =  0.540   2 corrector Newton steps
step  18  : stepsize = 0.2       lambda =  0.422   2 corrector Newton steps
step  19  : stepsize = 0.2       lambda =  0.296   2 corrector Newton steps
step  20  : stepsize = 0.2       lambda =  0.164   2 corrector Newton steps
step  21  : stepsize = 0.2       lambda =  0.027   2 corrector Newton steps
step  22  : stepsize = 0.0271    lambda =  0.000   2 corrector Newton steps
CPF TERMINATION: Traced full continuation curve in 22 continuation steps
>> results.cpf.events
ans =

  1x2 struct array containing the fields:

    k
    name
    idx
    msg

>> results.cpf.events(1)
ans =

  scalar structure containing the fields:

    k =  9
    name = QLIM
    idx =  1
    msg = gen 1 @ bus 1 reached 300 MVAr Qmax lim @ lambda = 0.9536 : bus 1 converted to PQ

>> results.cpf.events(2)
ans =

  scalar structure containing the fields:

    k =  22
    name = TARGET_LAM
    idx =  1
    msg = ZERO detected for TARGET_LAM event : CONTINUE
```

rdzman, 2020-12-04:
Slight correction ... the `idx` field is not the event index, but rather it provides additional information about the event. The fact that it is 1 for the first event in the case with Q limits, indicates that it is generator 1 that reached its Q limit.

abhyshr, 2020-12-04:
aah..thanks for the clarification. I suggest changing the message for TARGET_LAM to say "TARGET_LAM = <target_lam_value> reached : TERMINATE."

closed 2020-12-04 by rwl

rwl, 2020-12-04:
Thank you for explaining. The confusion arose from my failing to understand _multidimensional structs_. This page helped me find a solution:

https://octave.org/doc/v4.2.1/Structures-in-Oct_002dFiles.html