MATPOWER/matpower #110: no load information in matpower case

15151874182, 2020-12-17:
hello, i am new to matpower and electric, in case like 'case9.m', there are only bus,branch and generator data without load data,why?Any suggestion is appreciative.

rdzman, 2020-12-17:
Please see [`caseformat`](https://github.com/MATPOWER/matpower/blob/master/lib/caseformat.m) or Appendix B in the [MATPOWER User's Manual](https://matpower.org/docs/MATPOWER-manual-7.1.pdf) for a description of the MATPOWER case file format.

The MATPOWER case file format was based on earlier versions of other formats that included the load in the `bus` matrix. Active and reactive loads are specified in columns 3 and 4, respectively of the `bus` matrix.

15151874182, 2020-12-18:
> Please see [`caseformat`](https://github.com/MATPOWER/matpower/blob/master/lib/caseformat.m) or Appendix B in the [MATPOWER User's Manual](https://matpower.org/docs/MATPOWER-manual-7.1.pdf) for a [...]

thank u for ur answer!

closed 2020-12-18 by rdzman