MATPOWER/matpower #111: Error in results from power flow calculation

VictorOrmeno, 2021-01-17:
Dar Sirs--
I have been checking the results from running 'case9' and 'case4gs' that come with Matpower and have found a mismatch regarding the balance of Reactive Power. 
If you execute runpf('case4gs'), the balance of Reactive Power gives you:
Total reactive generation: 295.93 MVAR
Total reactive Load; 309.86 MVAR
Total reactive losses: 24.05 MVAR
This result is inconsistent. The same happens with runpf ('case9'). In this case you get:
Total reactive generation: 22.84 MVAR
Total reactive Load; 115.0 MVAR
Total reactive losses: 48.38 MVAR
Regards
Victor Ormeno
vormenos@gmail.com

rdzman, 2021-01-19:
The reactive losses you mention refer only to the series losses in the branches. For a full accounting of reactive power balance you need to include the line charging capacitance as well as any bus shunt elements.

Here's an example that includes all of the pieces.

```
>> r = runpf('case30', mpoption('out.all', 0, 'verbose', 0));
>> [loss, chg] = get_losses(r);
>> series_loss = imag(sum(loss));
>> chg_inj = sum(chg);
>> shunt_inj = sum(r.bus(:, VM).^2 .* r.bus(:, BS));
>> gen_q = sum(r.gen(:, QG));
>> load_q = sum(r.bus(:, QD));
>> gen_q + chg_inj + shunt_inj - load_q - series_loss

ans =

  -2.1622e-07

>> 
```

VictorOrmeno, 2021-01-20:
Thank you very much. Everything is clear now.

closed 2021-01-20 by VictorOrmeno