MATPOWER/matpower #114: Issue with test_matpower

standon41, 2021-02-22:
When I run test_matpower I get inconsistencies to whats given in the manual. I get a crashing error on 't_loadcase'. I would appreciate any information on this. 

The error is "Too many arguments"

standon41, 2021-02-22:
[image]

rdzman, 2021-02-25:
Make sure when you run the tests that your current working directory is a directory that you have permissions to write to, since the tests create some temporary files in it.

If that doesn't resolve the problem, could you please provide the output of the `mpver` command and of `t_loadcase`?

standon41, 2021-02-26:
Thank you for reaching out! Here you go:

>> mpver

MATPOWER               Version 7.1        08-Oct-2020
MATLAB                 Version 9.9        29-Jul-2020   Release: (R2020b)  
Optimization Toolbox   Version 9.0        29-Jul-2020   Release: (R2020b)  
MP-Test                Version 7.1        08-Oct-2020
MIPS                   Version 1.4        08-Oct-2020
MP-Opt-Model           Version 3.0        08-Oct-2020
MOST                   Version 1.1        08-Oct-2020
SDP_PF                 -- not installed --
SynGrid                Version 1.0.1      20-Jun-2019
Artelys Knitro         -- not installed --
BPMPD_MEX              -- not installed --
CLP                    -- not installed --
CPLEX                  -- not installed --
GLPK                   -- not installed --
Gurobi                 Version 9.1.0                 
IPOPT                  -- not installed --
MINOPF                 -- not installed --
MOSEK                  -- not installed --
OSQP                   -- not installed --
PARDISO                -- not installed --
PDIPMOPF               -- not installed --
SCPDIPMOPF             -- not installed --
SDPT3                  -- not installed --
SeDuMi                 -- not installed --
TRALMOPF               -- not installed --
YALMIP                 -- not installed --
Architecture:          MACI64

  MATPOWER 7.1 is distributed under the 3-clause BSD License.
  Please see the LICENSE file for details.

>> t_loadcase
1..240
Error using save
Too many input arguments.

Error in t_loadcase (line 40)
eval(['save ' matfile '.mat baseMVA bus gen branch areas gencost']);

rdzman, 2021-02-26:
What does it output when you type `which save`?

standon41, 2021-02-26:
>> which save
/Library/gurobi910/mac64/matlab/save.m

rdzman, 2021-02-26:
That's it. Apparently that `save.m` file (which is not part of my Gurobi install) is overriding Matlab's built-in save command. I'm not sure where that file came from, but you'll need to remove it from your Matlab path, or rename it or something in order for Matlab (and MATPOWER) to work properly.

standon41, 2021-02-26:
ok, I renamed it and I am still getting a crashing error on t_loadcase. What should be the value of "which save"

standon41, 2021-02-26:
I am assuming it is not this

 which save
built-in (/Applications/MATLAB_R2020b.app/toolbox/matlab/general/save)

rdzman, 2021-02-26:
Yes, that is the correct one. Now it looks like it's just the permission issue I mentioned first. Make sure you change your working directory to one you have write permissions for before running the tests.

closed 2021-02-26 by standon41

Chiman-AI, 2023-06-17:
[image]

Chiman-AI, 2023-06-17:
Would anyone be able to help me fix my errors?

rdzman, 2023-06-17:
What version of MATLAB are you using?

rdzman, 2023-06-17:
That looks like the issue introduced with MATLAB R2023a that was fixed with [this commit](https://github.com/MATPOWER/mp-opt-model/commit/478c42c9b74f8dc6dabeb5a073e4a1aa0196d7e8).

Cloning or downloading the latest master branch should fix it.

_Note that the GitHub repository does not include the `extras` directory with the contributed code from [MATPOWER Extras](https://github.com/MATPOWER/matpower-extras). So if you need those, you'll have to clone or download them separately._

Chiman-AI, 2023-06-17:
MatlabR2023a   and try with matpower7.1 and matpower8.0b, but got same error

Chiman-AI, 2023-06-17:
>  CAUTION: This email has originated from outside of the organisation. Do not click links or open attachments unless you have verified the sender and content is safe.
>
> (Btw, this is a different problem from the one originally posted in this issue, which was resolved a couple of years ago. In the future, please start a new issue rather than posting in an old, closed issue.)
>
> The error you are seeing was introduced with the release of MATLAB R2023a, after the release of MATPOWER 8.0b1, so you'll have to get the latest (unreleased) version of MATPOWER with the fix directly from GitHub. Go to the MATPOWER GitHub page<https://github.com/MATPOWER/matpower> and click on the green button that says "<> Code v" and either clone it or select "Download ZIP".
>
> That version should resolve the issue.

Hello,
I hope everything will go well,

Thank you for your answer.
It is my first time working with Matlab and Matpower.
For my MSc project, I need it.

Best Wishes,
Chiman Badri
 Aston AI student