MATPOWER/matpower #116: baseMVA changed in case33bw.m

TobiasRitschel, 2021-03-02:
From Matpower 7.0 to 7.1, baseMVA in case33bw.m has changed from 100 to 10. The remaining data appears to be unchanged.

Is this change in baseMVA correct?

rdzman, 2021-03-02:
This change was made for consistency with other distribution system cases that were added. Since the branch data are actually defined in the case file in Ohms and then converted to MW and MVAr programmatically, this change should not affect the solution, except possibly to improve numerical robustness.

If I recall, the original source did not specify a baseMVA, so I used a rule of thumb based on the total load to decide (see 9766ecf6).

TobiasRitschel, 2021-03-02:
Thank you for the quick answer. I will mark the issue as closed then.

closed 2021-03-02 by TobiasRitschel