MATPOWER/matpower #119: MATPOWER Version 6.0, 16-Dec-2016 -- AC Power Flow (Newton)  Newton's method power flow did not converge in 10 iterations.  >>>>  Did NOT converge (0.03 seconds)  <<<<<

Azmat1995, 2021-05-09:
Hello Everyone,
I'm working on IEEE-14 Bus system using MATPOWER.
when i was replace all the active load (Pd) in the given Bus data correctly and run the below code.
it shows: "AC Power Flow (Newton)  Newton's method power flow did not converge in 10 iterations".

What's the reason..?
 
function mpc = case5Fast_16
%CASE14    Power flow data for IEEE 14 bus test case.
%   Please see CASEFORMAT for details on the case file format.
%   This data was converted from IEEE Common Data Format
%   (ieee14cdf.txt) on 15-Oct-2014 by cdf2matp, rev. 2393
%   See end of file for warnings generated during conversion.
%
%   Converted from IEEE CDF file from:
%       http://www.ee.washington.edu/research/pstca/
% 
%  08/19/93 UW ARCHIVE           100.0  1962 W IEEE 14 Bus Test Case
%   MATPOWER

%% MATPOWER Case Format : Version 2
mpc.version = '2';

%%-----  Power Flow Data  -----%%
%% system MVA base
mpc.baseMVA = 100;

%% bus data
%	bus_i	     type	Pd	    Qd  	Gs     Bs  area	 Vm	     Va	  baseKV  	zone	Vmax	Vmin
mpc.bus = [
       1	           3	100	    0	    0	    0	1	0.5	      0	    0	       1    1.06  	0.94;
	2	           2	845.03	12.7	0	    0	1	1.045	-4.98	0	       1	1.06	0.94;
	3	           2	3367.53	19	    0	    0	1	1.01	-12.72	0	       1	1.06	0.94;
	4	           1	1741.13	-3.9	0	    0	1	1.019	-10.33	0	       1	1.06	0.94; 
	5	           1	360.93  1.6	    0	    0	1	1.02	-8.78	0	       1	1.06	0.94;
	6	           2	451.2	7.5	    0	    0	1	1.07	-14.22	0	       1	1.06	0.94;
	7	           1	100	    0	    0	    0	1	1.062	-13.37	0	       1	1.06	0.94;
	8	           2	100	    0	    0	    0	1	1.09	-13.36	0	       1	1.06	0.94;
	9	           1	1096.17	16.6	0   	19	1	1.056	-14.94	0	       1	1.06	0.94;
	10	           1	412	    5.8	    0	    0	1	1.051	-15.1	0	       1	1.06	0.94;
	11	           1	309.43  1.8	    0	    0	1	1.057	-14.79	0	       1	1.06	0.94;
	12             1	563.5   1.6	    0	    0	1	1.055	-15.07	0	       1	1.06	0.94;
	13	           1	478.23	5.8	    0	    0	1	1.05	-15.16	0	       1	1.06	0.94;
	14	           1	14.9	5	    0	    0	1	1.036	-16.04	0	       1	1.06	0.94;
];

%% generator data
%	bus	  PV	  Qg   Qmax	  Qmin  Vg  mBase	status	Pmax  Pmin	Pc1	  Pc2	Qc1min	Qc1max	Qc2min	Qc2max	ramp_agc   ramp_10	 ramp_30  ramp_q   apf
mpc.gen = [
	1	232.4	-16.9	10	   0	1	100	       1	332.4	0	 0	   0	   0	   0	  0	     0	      0	         0	       0	    0	    0;
	2	 40	     42.4	50	  -40	1	100	       1	140	    0	 0	   0	   0	   0	  0	     0	      0	         0	       0	    0	    0;
	3	 0	     23.4	40	   0	1	100	       1	100	    0	 0	   0	   0	   0	  0	     0	      0	         0	       0	    0	    0;
	6	 0	     12.2	24	  -6	1	100	       1	100	    0	 0	   0	   0	   0	  0	     0	      0	         0	       0	    0	    0;
	8	 0	     17.4	24	  -6	1	100        1	100	    0	 0	   0	   0	   0	  0	     0	      0      	 0	       0	    0	    0;
];

%% branch data
%	fbus	tbus	r	       x	    b	   rateA  rateB	     rateC	ratio  angle	status	angmin	angmax
mpc.branch = [
	1	    2	  0.01938	0.05917	  0.0528	0	    0	       0	   0	0	       1	-360	360;
	1	    5	  0.05403	0.22304	  0.0492	0	    0	       0	   0	0	       1	-360	360;
	2	    3	  0.04699	0.19797	  0.0438	0	    0	       0	   0	0	       1	-360	360;
	2	    4	  0.05811	0.17632	   0.034	0	    0	       0	   0	0	       1	-360	360;
	2	    5	  0.05695	0.17388	  0.0346	0	    0	       0	   0	0	       1	-360	360;
	3	    4	  0.06701	0.17103   0.0128	0	    0	       0	   0	0	       1	-360	360;
	4	    5	  0.01335	0.04211	    0	    0	    0	       0	   0	0	       1	-360	360;
	4	    7	   0	    0.20912	    0	    0	    0	       0	1.1 	0	       1	-360	360;
	4	    9	   0	    0.55618	    0	    0	    0	       0	1.1 	0	       1	-360	360;
	5	    6	   0	    0.25202	    0	    0	    0	       0	1.1 	0	       1	-360	360;
	6	   11	  0.09498	0.1989	    0	    0	    0	       0	   0	0	       1	-360	360;
	6	   12	  0.12291	0.25581	    0	    0	    0	       0	   0	0	       1	-360	360;
	6	   13	  0.06615	0.13027	    0	    0	    0	       0	   0	0	       1	-360	360;
	7	    8	   0	    0.17615	    0	    0	    0	       0	   0	0	       1	-360	360;
	7	    9	   0	    0.11001	    0	    0	    0	       0	   0	0	       1	-360	360;
	9	   10	  0.03181	0.0845	    0	    0	    0	       0	   0	0	       1	-360	360;
	9	   14	  0.12711	0.27038	    0	    0	    0	       0	   0	0	       1	-360	360;
	10	   11	  0.08205	0.19207	    0	    0	    0	       0	   0	0	       1	-360	360;
	12	   13	  0.22092	0.19988	    0	    0	    0	       0	   0	0	       1	-360	360;
	13	   14	  0.17093	0.34802	    0	    0	    0	       0	   0	0	       1	-360	360;
];

%%-----  OPF Data  -----%%
%% generator cost data
%	1	startup	    shutdown	n	x1	y1	...	xn	   yn
%	2	startup 	shutdown	n	c(n-1)	...	c0
mpc.gencost = [
	2	   0	     0	        3	0.0430292599	20	0;
	2	   0	     0	        3	0.25	        20	0;
	2	   0	     0	        3	0.01	        40	0;
	2	   0	     0	        3	0.01	        40	0;
	2	   0	     0	        3	0.01	        40	0;
];

%% bus names
mpc.bus_name = {
	'Bus 1     HV';
	'Bus 2     HV';
	'Bus 3     HV';
	'Bus 4     HV';
	'Bus 5     HV';
	'Bus 6     LV';
    
	'Bus 7     ZV';
	'Bus 8     TV';
	'Bus 9     LV';
	'Bus 10    LV';
	'Bus 11    LV';
	'Bus 12    LV';
	'Bus 13    LV';
	'Bus 14    LV';
};

% Warnings from cdf2matp conversion:
%
% ***** check the title format in the first line of the cdf file.
% ***** Qmax = Qmin at generator at bus    1 (Qmax set to Qmin + 10)
% ***** MVA limit of branch 1 - 2 not given, set to 0
% ***** MVA limit of branch 1 - 5 not given, set to 0
% ***** MVA limit of branch 2 - 3 not given, set to 0
% ***** MVA limit of branch 2 - 4 not given, set to 0
% ***** MVA limit of branch 2 - 5 not given, set to 0
% ***** MVA limit of branch 3 - 4 not given, set to 0
% ***** MVA limit of branch 4 - 5 not given, set to 0
% ***** MVA limit of branch 4 - 7 not given, set to 0
% ***** MVA limit of branch 4 - 9 not given, set to 0
% ***** MVA limit of branch 5 - 6 not given, set to 0
% ***** MVA limit of branch 6 - 11 not given, set to 0
% ***** MVA limit of branch 6 - 12 not given, set to 0
% ***** MVA limit of branch 6 - 13 not given, set to 0
% ***** MVA limit of branch 7 - 8 not given, set to 0
% ***** MVA limit of branch 7 - 9 not given, set to 0
% ***** MVA limit of branch 9 - 10 not given, set to 0
% ***** MVA limit of branch 9 - 14 not given, set to 0
% ***** MVA limit of branch 10 - 11 not given, set to 0
% ***** MVA limit of branch 12 - 13 not given, set to 0
% ***** MVA limit of branch 13 - 14 not given, set to 0

rdzman, 2021-05-10:
The first thing I always do with a new case is to use [`case_info()`](https://matpower.org/docs/ref/matpower7.1/lib/case_info.html) to see if there are any obvious data input errors.

In this case it shows that you have 9940.1 MW of load and only 272.4 MW of dispatched generation.

rdzman, 2021-05-10:
First, I'm not sure what you did to increase the generation capacity, but in the `case_info` results above, it still says the dispatched generation is set to 272.4, as in the original `case14`.

For a normal power flow problem the sum of the `PD` column in the `bus` matrix should be slightly less (because of losses) than the sum of the `PG` column in the `gen` matrix.

However, the load in your case is about 38 times larger than in the original `case14`! Even if you were to increase the generation by the same factor, there is no reason to expect the system to necessarily  converge when changing parameters by such huge amounts. That system will see a voltage collapse long before arriving at such extreme loading.

What is it that you are trying to do?

Azmat1995, 2021-05-11:
Sir actually I'm adding an EVs load to the bus, and see what effect it has on the system. 

Interm of voltage and losses.

Azmat1995, 2021-09-04:
Sir Actually, I'm trying to check the extra capacity of the given system
for that, I adding extra houses load to the installed capacity and then
check their response on each node.
But I'm facing a problem of convergence.

What should I do to solve this problem?

rdzman, 2021-09-04:
Sorry, I didn't realize this was not resolved.

You could use the continuation power flow to find out what the loading limit is on `case14`. There is also the [mx-maxloadlim] (https://github.com/MATPOWER/mx-maxloadlim) package included in `extras` that you can use to solve the same problem. My guess is that your non-converging power flows are simply a result of adding more load the that system can handle.

Azmat1995, 2021-09-05:
Yeah, you are right sir. But how to solve this issue which is due to
overloading of bus.

As there any method to solve the convergence problem...?

Thanks

rdzman, 2021-09-06:
In order to support larger loads, I suppose you need a system with higher voltages to be able to transmit the same amount of power with lower currents, lower losses and smaller voltage drops ... or you need transmission with lower impedances (e.g. more conductors in parallel).

If you are going to take an existing system (e.g. `case14`) and attempt to modify it to increase the load it can handle, it's going to be important for you to understand the relationships between the per-unit bases you are using and to have an idea what a parameter change corresponds to in terms of actual physical upgrades to your system ... assuming you need a plausible "upgrade" story to go with the modifications you're making.

Azmat1995, 2021-09-07:
Thanks for the tip! I really appreciate your help.
thanks a lot