MATPOWER/matpower #120: Consider a cloud CI service

acampbel, 2021-05-12:
Given public open source projects have free support for cloud ci services, consider leveraging one of them here. 🎉

https://blogs.mathworks.com/developer/2020/12/15/cloud-ci-services/

rdzman, 2021-05-13:
@acampbel, thanks for this heads up. I wasn't aware of these. We have been using CI with GNU Octave for over 4 years now, first with Travis CI and now GitHub Actions, but would love to have our code CI tested on MATLAB too.

Because all of our code must be tested on MATLAB and Octave, we use our own simple test framework, [MP-Test](https://github.com/MATPOWER/mptest), rather than MATLAB's built-in unit test framework.

Any chance you could provide a super simple example of a GitHub Action that runs a MATLAB script or function causing a  pass or fail based on whether some condition in the script/function is true or false?

And where do I find out more about the MATLAB environments provided with these free MATLAB CI services? For example, what operating systems are available? Which MATLAB versions? What toolboxes?

rdzman, 2021-05-13:
Looks like this should be helpful ...

   https://github.com/matlab-actions/overview/

acampbel, 2021-05-13:
Yes that should get you going. Also, here is another repo you might b=find interesting:

https://github.com/mathworks/ci-configuration-examples

> Looks like this should be helpful ...
> 
> https://github.com/matlab-actions/overview/

closed 2021-08-20 by rdzman