MATPOWER/matpower #121: issue with apply_changes when using MATLAB 2020a

michaelperl, 2021-07-12:
[bug_report.zip](https://github.com/MATPOWER/matpower/files/6801465/bug_report.zip)

When trying to use the function apply_changes using case9 and the chgtable in the comments, I get the following error:

> Index in position 1 exceeds array bounds (must not exceed 4).
> 
> Error in apply_changes (line 92)
>     change = num2cell(chgtab(kk(c), :));

The values of kk are: 

kk =
     1
     6
    11

I believe the issue is with the function find  

> "k = find(X) returns a vector containing the linear indices of each nonzero element in array X."

apply_changes (line 79): `kk = find(label == chgtab(:, CT_LABEL));`

I believe based on the documentation of find 

> [row,col] = find(___)

The code can be changed as follows to fix the bug:

apply_changes (line 79): `[kk,~] = find(label == chgtab(:, CT_LABEL));`

Attached is a script to reproduce the bug.

rdzman, 2021-07-12:
Thanks. It turns out that `apply_changes` expects `label` to be a scalar. It's not implemented for vector values and should actually throw a useful error message in that case.

You can however, have multiple changes in `chgtable` with the same label in order to apply multiple changes at once.

michaelperl, 2021-07-14:
Thank you for the quick response. Using the same label for multiple lines works for me.

closed 2021-07-14 by michaelperl