MATPOWER/matpower #124: How to use connected_component function on case-file ?

eendru, 2021-09-03:
What's a simplest way to use connected_components() function:

```
mpc = loadcase(case4gs)
connected components (???) 
```

Thanks.

rdzman, 2021-09-03:
The `connected_components()` function is not meant to be used directly (think of it more as a private function). The public functions that use it are [`case_info()`](https://matpower.org/docs/ref/matpower7.1/lib/case_info.html), [`find_islands()`](https://matpower.org/docs/ref/matpower7.1/lib/find_islands.html), and [`extract_islands()`](https://matpower.org/docs/ref/matpower7.1/lib/extract_islands.html).

You're probably looking for one of the latter two.

You can mark a load bus as out-of-service by setting the `BUS_TYPE` column (col 2) of the `bus` matrix to `NONE` (4).
You can mark branch as out-of-service by setting the `BR_STATUS` column (col 11) of the `branch` matrix to 0.
If you do both, they should both be eliminated from the power flow. If not, provide an example and I'll look into it.

Taking a generator off-line does not change the load ... the difference will just be taken up by the slack generator, but the load should remain the same.

Hope this helps.

eendru, 2021-09-06:
ok, thank you.
I will check.

closed 2021-09-06 by eendru