MATPOWER/matpower #125: Octave 6.3.0 test fails

thomsonian2023, 2021-09-24:
I am using Octave 6.3.0 on windows. I get the following error:

t_cpf...................error: r(10): out of bound 9 (dimensions are 1x9)
error: called from
    t_cpf at line 556 column 9
    t_run_tests at line 61 column 5
    test_matpower at line 173 column 8
    install_matpower at line 392 column 13](url)

rdzman, 2021-09-24:
And this is with MATPOWER 7.1?

thomsonian2023, 2021-09-24:
Yes Version 7.1

rdzman, 2021-09-24:
Hmmm ... I am not able to replicate the error on Octave 6.3.0 on Windows with MATPOWER 7.1. All tests pass fine here.

Could you try the following and let me know what you find?

1. Confirm that `t_cpf` runs without error on a previous version of Octave.
2. If so, type `which t_cpf` in both versions of Octave to ensure that both are using the correct file from the same installation of MATPOWER 7.1.

If neither of those give any clues, my next suggestion would be to try reinstalling a fresh download of MATPOWER, in case a file in the installation got inadvertently modified somehow.

closed 2021-09-26 by thomsonian2023