MATPOWER/matpower #126: Error reading RAW file

thomsonian2023, 2021-09-24:
My code is:

fn = 'X:\z2.RAW'
mpc = psse2mpc(fn);     %%%%%% Converts PSSE/RAW File to MATPOWER Case

I am using Windows OS, Matpower 7.1 and Octave 6.2.0. I get the following error:

Splitting into individual lines ...................... 44 lines read ... done.
Analyzing sections ..................................... 15 sections ... done.
Attempting to determine PSS/E revision from content.
Parsing case identification data ......... unknown rev, using rev 23 ... done.
Parsing      3 lines of bus data .......................................error: cellfun: all values must be
scalars when UniformOutput = true
error: called from
    psse_parse_section at line 209 column 32
    psse_parse at line 155 column 23
    psse2mpc at line 71 column 18
    shift_factor at line 4 column 5

I am using the following RAW file. Please note that my real life RAW file is much larger and is impossible to paste here. So, I have deleted intermediate lines from the RAW file. I get the same error when I am using either the full or abridged RAW file. 

#---------------------------Z2.RAW---------------------
0    100 /PSS�E 30      Wed Feb 17 11:29:28 2021
PSS�E v30 RAW data export from PSS/O
20210217110948
1,'asdwer  ',138,1,0,0,11,38,1,0,115,    /*[1 asdwer ]*/
2,'asdsdfgfd     ',138,1,0,0,11,38,1,0,115,    /*[2 asdsdfgfd ]*/
4,'whatsdsd        ',138,1,0,0,11,38,1,0,115,    /*[4 whatsdsd ]*/
0 / END OF BUS DATA, BEGIN LOAD DATA
1,'1 ',1,11,38,4.65,0.142,0,0,0,0,115,    /*[1 asdwer 1  LD]*/
4,'2 ',1,11,38,1.803,0.212,0,0,0,0,115,    /*[4 whatsdsd 2  LD]*/
4,'1 ',1,11,38,1.803,0.212,0,0,0,0,115,    /*[4 whatsdsd 1  LD]*/
0 / END OF LOAD DATA, BEGIN GENERATOR DATA
40426,'B1',0,0,20,-20,1,0,137,0,0.00001,0,0,1,0,100,137,0,456,1,0,1,0,1,0,1,    /*[BL_CS_BLT]*/
80169,'EQ',0,0,40,-40,0.97826,0,110,0,0.00069,0,0,1,1,100,110,0,456,1,0,1,0,1,0,1,    /*[LA_GEN]*/
0 / END OF GENERATOR DATA, BEGIN BRANCH DATA
1,5,'1 ',0.002788,0.0149,0.004151,259,259,259,0,0,0,0,1,3.82,115,1,0,1,0,1,0,1,     /*[YSW 1 ]*/
2,4,'1 ',0.004453,0.02379,0.006649,259,259,259,0,0,0,0,1,6.1,115,1,0,1,0,1,0,1,     /*[d 1 ]*/
0 / END OF BRANCH DATA, BEGIN TRANSFORMER DATA
36,35,0,'1 ',1,1,1,0,0,2,'NAT1    ',1,115,1,0,0,0,0,0,0
0.0055501,0.12595,100
1,69,0,100,100,100,1,36,1.1,0.9,1.04283,1.00283,33,0,0,0,    /*[35 HEARNE8 138 36 HEARNE9 69 1 ]*/
0.975,138
0 / END OF TRANSFORMER DATA, BEGIN AREA DATA
1,0,0,10,'whatsdsd2    '
2,0,0,10,'RSP    '
3,0,0,10,'TEP    '
0 / END OF AREA DATA, BEGIN TWO-TERMINAL DC DATA
0 / END OF TWO-TERMINAL DC DATA, BEGIN VSC DC LINE DATA
0 / END OF VSC DC LINE DATA, BEGIN SWITCHED SHUNT DATA
99,1,1.03068,0.99068,0,100,'        ',7.5,1,7.5,    /*[99]*/
133,1,1.04857,1.00857,0,100,'        ',16,2,8,    /*[133SWS]*/
0 / END OF SWITCHED SHUNT DATA, BEGIN IMPEDANCE CORRECTION DATA
0 / END OF IMPEDANCE CORRECTION DATA, BEGIN MULTI-TERMINAL DC DATA
0  /END OF MULTI-TERMINAL DC DATA, BEGIN MULTI-SECTION LINE DATA
0 / END OF MULTI-SECTION LINE DATA, BEGIN ZONE DATA
1,'TEMPORARY   ',    /*[TEMARY]*/
2,'O_TAYLOR    ',    /*[O_TAR]*/
3,'DENTON      ',    /*[DTON]*/
0 / END OF OWNER DATA, BEGIN FACTS DEVICE DATA
0 / END OF FACTS DEVICE DATA

rdzman, 2021-09-24:
My best guess is that the text encoding of the RAW file is preventing MATPOWER from correctly detecting the PSS/E version automatically. It may work if you specify it as version 30 explicitly, like this ...
```
mpc = psse2mpc(fn, 0, 30);
```

And, if you like, you can send me the RAW file by direct e-mail and I'll see if I can confirm my guess and possibly make `psse2mpc` more robust to different text encodings.

thomsonian2023, 2021-09-24:
Yes that is what the problem was.

closed 2021-09-24 by thomsonian2023