MATPOWER/matpower #128: psse2mpc is not reading the branch name

thomsonian2023, 2021-09-26:
I am using the following code on Windows 10, Matpower 7.1, Octave 6.2.0:

clear; clc; 
fn = 'X:\z1.RAW'
mpc = ext2int(psse2mpc(fn, 0, 30));     %%%%%% Converts PSSE/RAW File to MATPOWER Case
H = makePTDF(mpc);    %%%%% Calculation of Shift Factor Matrix

When I look into the variable Editor, it shows only the mpc.branch of type double but no mpc.branchname.

For bus it has both mpc.bus and mpc.busname

In the RAW file, the branch data is of the following format:

0 / END OF GENERATOR DATA, BEGIN BRANCH DATA
1,5,'1 ',0.002788,0.0149,0.004151,259,259,259,0,0,0,0,1,3.82,115,1,0,1,0,1,0,1,     /*[1 RW 1 ]*/
1,964,'1 ',0.008969,0.046064,0.012754,155,170,196,0,0,0,0,1,11.04,115,1,1033,1,0,1,0,1,     /*[1 REK 1 ]*/
2,4,'1 ',0.004453,0.02379,0.006649,259,259,259,0,0,0,0,1,6.1,115,1,0,1,0,1,0,1,     /*[2 KA 1 ]*/

rdzman, 2021-09-27:
This is a limitation in the current MATPOWER case file format, which does not include branch names.

While bus names are also not part of the official case format described in the manual, at some point I became aware that multiple people were adding them to a cell array in `mpc` so I decided to standardize the field name to `busnames` and the handling of those bus names was added as an undocumented feature in a few places, including import from PSS/E RAW files.

If you are interested in adding branch name support, I'd be happy to consider including any contributed code for doing so. Some of my current MATPOWER work will eventually result in some major updates to the MATPOWER case file format, at which time support for branch names will be included.

thomsonian2023, 2021-09-27:
I will share codes if I am able to figure out the regex in Octave. Thanks for answering questions.

closed 2021-09-27 by thomsonian2023