MATPOWER/matpower #129: Relative gen vs load bus from RAW file

thomsonian2023, 2021-09-27:
Starting from a RAW file is it possible to use Matpower to understand which bus is a relative gen bus and which is a relative load bus when two random bus are selected. 

e.g. if I select two random bus one with generation G1 and another load bus L1, it is possible that flow is from L1 to G1, so in this case it can be said that L1 is relative generation node compared to bus G1. Question is, is there a mathematical way to figure this relationship using Matpower?

rdzman, 2021-09-27:
I'm afraid I don't quite understand the question.

Once you have a power flow (or OPF) solution, you can certainly determine the flow on any given branch or determine the net power injection at any bus.

Could you clarify what you mean by "relative generation node" or "relative load node"? And what do you mean by "flow is from L1 to G1" ... on a specific branch explicitly connecting the two, or something else?

thomsonian2023, 2021-09-27:
In a PJM like ISO where lets say we have bus A and B. If we fundamentally know that bus A has more generation nearby and bus B has more load nearby, the power will flow from A to B even if not directly connected. I was wondering whether there was a way to make such statement more quantitatively.

rdzman, 2021-09-27:
It sounds like you are looking for something like a "local net injection index" (my term). I'm not aware of such an index, but MATPOWER could certainly be helpful in computing it once it is precisely defined.

It seems to me that all you need is a useful definition of "nearby" and a rule for computing, for each node, some sort of weighted average of nodal net injections of "nearby nodes". So I think the issue is coming up with a precise definition of the index that would provide you with the information you are looking for.

closed 2021-09-28 by thomsonian2023