MATPOWER/matpower #131: Shift factor for transformers

thomsonian2023, 2021-11-11:
I used the following code:

fn = 'myFile.RAW'
mpc = ext2int(psse2mpc(fn, 0, 30));     %%%%%% Converts PSSE/RAW File to MATPOWER Case
H = makePTDF(mpc);    %%%%% Calculation of Shift Factor Matrix

It generates shift factor for branches but not for transformers. What do I modify in the above code to get the shift factor for transformers ?

rdzman, 2021-11-11:
The transmission lines and transformers in the RAW file both end up in the `branch` matrix in the current MATPOWER (lines first, then transformers), so it should simply be a matter of locating which rows in `branch` correspond to the transformers.

thomsonian2023, 2021-11-11:
In my RAW file between BEGIN BRANCH DATA and END BRANCH DATA, there are number of branches which are not showing up in the variable mpc.branch. I think that is what made me think maybe it is missing some transformer shift factors. Is there a reason not all branches show up in the Shift factor matrix ?

rdzman, 2021-11-11:
Well, the `ext2int()` conversion removes all off-line branches, so maybe it's just the off-line branches that are missing?

If that's not it, maybe you'd get some clues by looking at the verbose output from `psse2mpc`.
```
mpc = psse2mpc('myFile.RAW', 1, 30);
```
Or you could also save the MATPOWER case with ...
```
savecase('my_matpower_case', mpc);
```
... and open both the RAW and MATPOWER case files in a text editor and see if you can find which branches are missing.

thomsonian2023, 2021-11-12:
Feels like ext2int is causing the missing data. 

E.g. 
The following transformer is missing:
1,138
39530,39531,0,'2 ',1,1,1,0,0,2,'WCL2    ',1,1050,1,0,0,0,0,0,0
0.026795,0.63705,25
1,138,0,25,25,25,-1,39530,1.05,0.95,1.025,0.975,5,0,0,0,    /*[39530 asdsdfgdfgdf 138 39531 sdfsfdsd12.47 2 ]*/
1.00625,12.47
39561,39560,0,'1 ',1,1,1,0,0,2,'ANG1    ',1,1050,1,0,0,0,0,0,0
0.024006,0.47255,47
1.0425,13,0,47,47,47,1,39561,1.14675,0.93825,1.0332,0.9668,33,0,0,0,    /*[39560 sfdfds138 39561 asfsds 12.47 1 ]*/

rdzman, 2021-11-12:
The transformer that connects 39530 to 39531 (and the one that connects 39561 to 39560) appears to have a status of 1 (in-service) so `ext2int` should not be removing them. I wonder whether the comments at the end of some of the lines are handled correctly by my RAW parser.

Any chance you could send me the RAW file (via direct e-mail) for testing?

rdzman, 2021-11-12:
Thanks for sending the file.

I'm not any problems. 

The original MATPOWER case (`mpc` below) has the expected number of rows in the `branch` matrix (9182 branches + 2182 transformers = 11364) and the transformer you listed appears at row 10247.

The internally numbered case (`mpci` below), after bus number conversion and removal of off-line equipment using `ext2int()`, still has that transformer present, now at row 9145.

```
>> define_constants
>> mpc = psse2mpc('Network.RAW', 1, 30)
Reading file 'Network.RAW' ............................................. done.
Splitting into individual lines ................... 35441 lines read ... done.
Analyzing sections ..................................... 17 sections ... done.
Forcing interpretation as PSS/E revision 30
Parsing case identification data ......... unknown rev, using rev 30 ... done.
Parsing   9131 lines of bus data ....................................... done.
Parsing   5336 lines of load data ...................................... done.
Parsing   1019 lines of generator data ................................. done.
Parsing   9182 lines of branch data .................................... done.
Analyzing transformer types ...........  2182(0) two(three)-winding. ... done.
Parsing   2182 lines of 2-winding transformers (1) data ................ done.
Parsing      0 lines of 3-winding transformers (1) data ................ done.
Parsing   2182 lines of 2-winding transformers (2) data ................ done.
Parsing   2182 lines of 2-winding transformers (3) data ................ done.
Parsing   2182 lines of 2-winding transformers (4) data ................ done.
Parsing      0 lines of 3-winding transformers (2) data ................ done.
Parsing      0 lines of 3-winding transformers (3) data ................ done.
Parsing      0 lines of 3-winding transformers (4) data ................ done.
Parsing      0 lines of 3-winding transformers (5) data ................ done.
Parsing     64 lines of area data ...................................... done.
Parsing      0 lines of two-terminal DC (1) data ....................... done.
Parsing      0 lines of two-terminal DC (2) data ....................... done.
Parsing      0 lines of two-terminal DC (3) data ....................... done.
-----  WARNING:  Found section labeled:    'VSC DC LINE'
-----            Expected section labeled: 'VOLTAGE SOURCE CONVERTER'
Parsing    968 lines of switched shunt data ............................ done.
Skipping   565 lines of zone data ...................................... done.
Skipping   428 lines of owner data ..................................... done.
-----  WARNING:  Found section labeled:    'FACTS DEVICE'
-----            Expected section labeled: 'FACTS CONTROL DEVICE'
WARNING: No bus voltage magnitude limits provided.
         Using defaults: VMIN = 0.9 p.u., VMAX = 1.1 p.u.

mpc = 

  struct with fields:

     baseMVA: 100
         bus: [9131×13 double]
    bus_name: {9131×1 cell}
      branch: [11364×13 double]
         gen: [1019×21 double]

>> k = find(mpc.branch(:, F_BUS) == 39530 & mpc.branch(:, T_BUS) == 39531)

k =

       10247

>> mpci = ext2int(mpc)

mpci = 

  struct with fields:

     baseMVA: 100
         bus: [8851×13 double]
    bus_name: {8851×1 cell}
      branch: [10205×13 double]
         gen: [990×21 double]
       order: [1×1 struct]

>> e2i = mpci.order.bus.e2i;
>> i = find(mpci.branch(:, F_BUS) == e2i(39530) & mpci.branch(:, T_BUS) == e2i(39531))

i =

        9145

>> 
```

You may find it useful to know that `mpci.order.branch.status.on` contains a vector (corresponding to the rows of `mpci.branch`) with the indexes to the respective rows of `mpc.branch`. Similarly, indices of the off-line rows of `mpc.branch` are available in `mpci.order.branch.status.off`.

```
>> mpci.order.branch.status

ans = 

  struct with fields:

     on: [10205×1 double]
    off: [1159×1 double]

>> i = find(mpci.order.branch.status.on == k)

i =

        9145
```

Hope this helps.

closed 2021-11-12 by thomsonian2023